Beckhoff/ADS #4: can the PLC send an "unsolicited" message

delarosa1962, 2015-09-25:
I see the bit for request|response in the header. As a client it is clear to set as request --- when the reply comes the response bit should be set. So, in the event that the PLC decides to send a block of data --- what would the instruction be in the PLC and what should the header/command would look like at the client when the message is received ?

soberschmidt, 2015-09-25:
It is a client API actually without server functions. Use notification to update your client without polling

delarosa1962, 2015-09-25:
Hello,

So, if the SoftPLC has a UDT that represents some transaction and the PLC is updating it --- would the PLC be able to issue a "send UDT now" ? I understood that the notifications can be used, this is fine. But how are notifications handled for an UDT or array for example and their updates are not done in one scan for example. Other PLC systems have instructions to send data (ie, Siemens = TSEND, AG_SEND, Rockwell = MSG, Mitsubishi = BUFSND)

soberschmidt, 2015-10-02:
Notifications can be used for standard datatypes (bool, int, etc) and complex datatypes (structs, arrays....). Their is no difference

delarosa1962, 2015-10-02:
Thanks. Is the notification sent because the data changes or because the SoftPLC decidest to send it?

soberschmidt, 2015-10-02:
depend of the attributs. the plc can sent it cyclic or onchange

delarosa1962, 2015-10-02:
What PLC instruction will deliver a block of data ?

soberschmidt, 2015-10-03:
check your issue #3

closed 2015-10-03 by soberschmidt