Beckhoff/ADS #9: AdsNotificationAttrib

Satheesh-asml, 2015-10-08:
Hi,
Could you please help me to understand the event notification related queries

AdsNotificationAttrib::cbLength : Length of the data that is to be passed to the callback function.

What would be the length of cbLength data of the AdsNotificationAttrib structure variable passed as a
parameter to register a notification via a call to the AdsSyncAddDeviceNotificationReq 

is the related to the 

AdsSyncAddDeviceNotificationReq::hUser parameter
[in] 32-bit value that is passed to the callback function.
if yes, it would be 4 bytes value always right?

So, What kind of data would be passed back to the callback function via pNotification->data and what would be the size of the data?

How to handle if i require more than 550 notification?

nCycleTime
The ADS server checks whether the variable has changed after this time interval. The unit is 100 ns.

What would be the time delay if i registered for 100 notification with the nCycleTime set to 100ns?

soberschmidt, 2015-10-09:
Please check the spec: http://infosys.beckhoff.de/index.php?content=../content/1033/tcadsdll2/html/tcadsdll_funcadssyncadddevicenotificationreq.htm&id=21801

The limitation is only related to our TcAdsDll for windows. But keep in mind that notifications cost time. ADS isn't fieldbus. The delay isn't predictable. But you get an exact timestamp by the notification.

Satheesh-satz, 2015-10-09:
Hi Sven,

Sorry about not being clear, My questions are ...

Question 1:
From the spec : http://infosys.beckhoff.com/english.php?content=../content/1033/tcadsdll2/html/TcAdsDll_StrucAmsAddr.htm&id= 

The hUser parameter to the function "AdsSyncAddDeviceNotificationReq" says 
hUser
[in] 32-bit value that is passed to the callback function.

I understand that from the sample provided at the following link 

http://infosys.beckhoff.com/english.php?content=../content/1033/tcsample_vc/html/tcadsdll_api_cpp_sample08.htm&id=

That it is the handle to the plc variable name, but my question here is what if i want to use the index group and offset value instead of passing the handle to the variable name.

Question 2:
And the spec : http://infosys.beckhoff.com/english.php?content=../content/1033/tcadsdll2/html/TcAdsDll_StrucAdsNotificationAttrib.htm&id=

The cbLength AdsNotificationAttrib structure member variable says

cbLength
Length of the data that is to be passed to the callback function.

I understand the callback function will be called with "AdsNotificationHeader* pNotification" and the pNotification->data will be have the change in value. is that right?

If yes, what if i set cbLength to a value of 12 or more than 4 bytes, how does this affect the data passed back to the call back function.

soberschmidt, 2015-10-14:
Q1: Indexgroup and Offset can also be used to announce a notification
Q2: Our TcAdsDll.dll for Windows OSs differs to our OpenSource AdsLib to provide a wider range of architecture support. Please use the structure information of our AdsLib.

closed 2015-12-18 by soberschmidt