Beckhoff/ADS #14: Error 1861

Satheesh-satz, 2015-10-27:
Hi Sven,

I am getting the error 1861 in one my linux machine, I did turn off the firewall and all the route has been added on the twincat service.

Could you help me more to understand and debug this issue.

1861: timeout elapsed "Check ADS routes of sender and receiver and your  firewall setting"

Thx

Ramgopalcr, 2018-02-02:
In fact it was working some time back not sure if there is something from network which I can check, but why twincat-3 direct fb for ads read working

pbruenn, 2018-02-02:
Yes, your issues sounds related. I will answer here and close #56 as duplicate.

Do you use `AdsSetLocalAddress()` to set your AdsLib AmsNetId?
https://github.com/Beckhoff/ADS/blob/56bc355c7eeb38bb4400b939037a256984742ded/AdsLib/AdsLib.h#L71

Check which AmsNetId your local TwinCAT router would use when you connect using "direct-fb"(TcAdsDll.dll). Either look into the routing table of your TwinCAT server or check the router setting on the client like this:
 
[image]

Then before you open any AdsPort in your AdsLib example. Call `AdsSetLocalAddress()` with that AmsNetId
```
diff --git a/example/example.cpp b/example/example.cpp
index fe893a2..833fa4c 100644
--- a/example/example.cpp
+++ b/example/example.cpp
@@ -180,6 +180,8 @@ void runExample(std::ostream& out)
     static const AmsNetId remoteNetId { 192, 168, 0, 231, 1, 1 };
     static const char remoteIpV4[] = "ads-server";

+    AdsSetLocalAddress({192, 168, 0, 114, 1, 1});
+
     // add local route to your EtherCAT Master
     if (AdsAddRoute(remoteNetId, remoteIpV4)) {
         out << "Adding ADS route failed, did you specified valid addresses?\n";
```

For more background see commit c1bfedd25988468ac0f46e9cf0fff4d4be3ffa52 and let me know if this didn't fix your issue.

Ramgopalcr, 2018-02-02:
Hi 

Thanks and it did work and resolved my issue.

One more query on the general stuff, is we want to basically integrate it in to our WinCE machine (client is WinCE 6.0) and communicate with the CX device.

But in order to add the static route, is there any other alternative other than installing Twincat on the CE machine, like some software which is not run time but just to add the route.At the end, this static route is a redundant operation as we have the AdsAddroute in some sense.

pbruenn, 2018-02-02:
Some background on "ADS routing":
With TwinCAT installed on your system you get a full blown ADS router. Ignoring a lot of details that's just a TCP/IP multiplexer. Having a machine A (your WinCE client) and a machine B (your TwinCAT CX) which want to communicate with ADS. This is what happens Behind the Scenes:
Your machine A opens a TCP connection to machine B. Over this TCP connection multiple "ADS connections" can be multiplexed. If you have different ADS servers running on your TwinCAT machine. You will have multiple AmsNetIds there. So the component on machine B listening to the TCP socket will need to inspect the incoming ADS packet for its Destination AmsNetId. That component is the TwinCAT ADS router. When the server on machine B wants to reply to a response from machine A. It doesn't know anything about TCP/IP. All it can do is give the response to the TwinCAT ADS router and tell him the destination AmsNetId. The router will then lookup the destination IP in it's routing table. If you would use TwinCAT on machine A. It is very likely that entry was already semi-automatically added. If you don't have TwinCAT on machine A. You need to setup the static route on machine B manually. All tools you need, should be installed already.
What we skipped is, how does the first message from your example arrive at machine B's ADS router? AdsLib was intended to be used for simple but portable (non Windows) applications. That's why there is no real ADS router implemented. All we have is a simple translation table AmsNetId -> IP address. That table is controlled by `AdsAddRoute()`. Another thing missing is the local AmsNetId. On TwinCAT systems that AmsNetId is controlled by the router. As we don't have a real router, we derive the AmsNetId from our local IP. That mechanism is not very deterministic, so `AdsSetLocalAddress()` was introduced.
1. Make sure to assign a deterministic AmsNetId on your AdsLib client -> `AdsSetLocalAddress()` 
2. In your AdsLib client add a route to your TwinCAT system -> ´AdsAddRoute()´
3. On your TwinCAT system add a route for your AdsLib client

closed 2018-11-13 by pbruenn