Beckhoff/ADS #15: Notification example does not report updated values

towillem, 2015-10-27:
I'm testing the notification mechanism from the ADS protocol using the AdsSyncAddDeviceNotificationReqEx call.
The provided example has been build on Ubuntu 14.04 LTS and Windows7-64bit cygwin64.
On both platform the ADS client runs without errors.
The application data from the PLC-TestProject in the array is continously updated.
The reported value from the ADS client for indexgroup 0x4020 with offset 4 is always 0 on both plaforms, that is incorrect. Even with a sample size of 1, the byte value should be update conform the value in the array.
Check the attached screenshots of the PLC application and the reported output of the ADS clients on both platforms.
The screenshot of the About TwinCAT System is also attached.

[image]
[image]
[image]
[image]
[image]

soberschmidt, 2015-10-28:
Hi Ton,

please verify the adresses with our target browser (VS Menue/Scope/TargetBrowser). Furthermore provide WireShark Dump via Mail.

closed 2015-10-28 by soberschmidt

Satheesh-satz, 2015-10-28:
Hi Sven,

I am seeing an error saying "Symbol could not be found" when i tried to open the either the "plcTask" or "Port_851" sub items listed on my system.

Could you please help me with that .

towillem, 2015-10-29:
Hi Sven,

Yep that’s it, the address from the example is not in sync with the PLC Test project.
After updating the address in the example, the values are updated.

regards, Ton

-- The information contained in this communication and any attachments is confidential and may be privileged, and is for the sole use of the intended recipient(s). Any unauthorized review, use, disclosure or distribution is prohibited. Unless explicitly stated otherwise in the body of this communication or the attachment thereto (if any), the information is provided on an AS-IS basis without any express or implied warranties or liabilities. To the extent you are relying on this information, you are doing so at your own risk. If you are not the intended recipient, please notify the sender immediately by replying to this message and destroy all copies of this message and any attachments. The sender nor the company/group of companies he or she represents shall be liable for the proper and complete transmission of the information contained in this communication, or for any delay in its receipt.