Beckhoff/ADS #22: Sum commands support

jflbr, 2016-05-04:
Hi
I don't think that opening an issue was the right choice for asking questions about the lib, sorry if I did wrong. 

After testing the reading/writing/ features, I found myself stuck while trying to use notification feature with a "non-memory" variable. In the lib example we subscribe to a variable declared as follows:
```
byByte		AT %MB0 : ARRAY [0..1027] OF BYTE;
```
It works fine.
But when I adapt the example with a var declared as below:
```
monitoredOutput  AT %Q*: BOOL;
```
It doesn't work and I get a `0x701` error code (ie. Service is not supported by server). Note the * char. I don't have physical I/O to address. 

I'm using TwinCAT 3 and I got the Index group/offeset from the variable detail view. For example, inside of the ADS Info field I had an Index group of 0x1010010 and a Index offset of 0x8107D209. The same field specifies a port of 350 which is different from the one (851) the TwinCAT PLC listens on. I even thought that my problem came from that so I tried to use that  port (350) as a parameter to the `AdsSyncAddDeviceNotificationReqEx` function. It failed.

So my question is does the `AdsSyncAddDeviceNotificationReqEx` function support variables declared with the % character?

Best regards,
Jeyfel B

soberschmidt, 2016-05-04:
The issue is not related to the lib. TwinCAT do not actually provide notifications for IO-Tasks. A mapping to the PLC could be a workaround.

Regards
Sven

closed 2016-05-04 by soberschmidt

jflbr, 2016-05-18:
Thank you for suggesting the workaroud.
However, I still have a `0x701` (service is not supported by server) error even after mapping my PLC variables to an EtherCAT device (simulation) I/O variables. Once again, I admit this is not the place to ask my question. Indeed, I tried to access the EtherCAT device (simulation) variables from the test program provided with TwinCat. The error is the same. So this is defenitely not related to the lib, like my previous question. So before I go further for nothing, does the EtherCAT simulation support ADS commands?

Just for the record, I'm student working on a virtual commissioning project for my internship.

Regards
Jeyfel B