Beckhoff/ADS #26: Suggestion: Add additional interfaces to the library

xift, 2016-07-05:
Hi,
thanks for making this open source. I work with TwinCAT daily and was happy to see some of the implementation. The library code is pretty clean as well. Nice work!

So my suggestion would be to build upon this library and add interfaces.

pbruenn, 2016-07-07:
Hi Michael,
yes I think it's a good idea to extend this lib with an object oriented interface. What's holding me from doing it, is my lack of a concrete use case. I don't want to implement something without a prove of it's real usage.
So my suggestion would be, implement it on a branch of your forked repository and let us know. Then we have a base on which we can discuss the integration into upstream.

Best regards,
Patrick

xift, 2016-07-07:
I agree that there probably is no real use case for Beckhoff in a new interface. 
However, the API is a bit old-fashioned and uncomfortable to use (Handle here, error code there, ...). 
I accept the challenge and see what I can do. I actually stumbled upon this repository when looking for a side-project that might be beneficial at work. So this might be a perfect opportunity.

Michael

xift, 2016-08-02:
Gave it a shot...
#35

closed 2016-08-02 by xift