Beckhoff/ADS #32: non-GUI way to add remotes to the router table possible?

rockonedege, 2016-07-30:
From all the documentation I read, they all reference to some GUI interface to add a remote to the router table. is there any API or steps  that can be automated programably?

If this is possible,I'd like to write a small stub program on the TWinCAT server helping this adslib to register itself before any ADS communication.

soberschmidt, 2016-08-01:
It is possible to add routes via ADS (check TcUtilities.lib for interface information). But for that you need an existing route.

rockonedege, 2016-08-01:
Can you please provide a link? I could not locate it with a google search. Here's what I am planning to do now: I script a small program listening to a specific tcp port, once it receives a request, it writes to the OS registry the remote ip and restarts TWINCAT. By looking at the system manager Gui, it seems being added. But I have yet to verify it works or not.

soberschmidt, 2016-08-03:
Thats the link to the [TcUtilities.lib](https://infosys.beckhoff.com/index.php?content=../content/1031/tcplclibutilities/html/tcplclibutilities_overview.htm&id=). 

Remind that you have to restart TwinCAT if you change the route settings in the registry. If you want to avoid the route handling, use the [AdsWebService](http://infosys.beckhoff.de/index.php?content=../content/1031/tcsample_tcadswebservicejs/html/intro.html)

closed 2017-01-05 by pbruenn