Beckhoff/ADS #33: ADS read error example: 1861

reyanvaldes, 2016-08-01:
Hi, I am trying to run the example of ADS library but I am getting same error " ADS read error example: 1861", probably I am doing something wrong. It looks it get connection to the ADS server v2.11. I developed a tiny program running on TwinCATon Windows XP with  IP:192.168.10.116. I changed the code in runExample:
    static const AmsNetId remoteNetId {"192.168.10.116.1.1"}; // { 192, 168, 10, 54, 1, 1 };
    static const char remoteIpV4[] = "192.168.10.116";
    ....
    const AmsAddr remote { remoteNetId, 801 }; 
I created on TwinCAT very simple project with some DIs with  address 0.0-0.7 (Port: 801, IGrp: 0xF021, IOffs:0x0, Len:1) and DI -Byte with address 0 (Port: 801, IGrp: 0xF020, IOffs:0x1, Len:1)

In the readExample I changed:
  const long status = AdsSyncReadReqEx2(port, &server, 0xF020, 0, 1, &buffer, &bytesRead);

But I am getting the error mentioned all time. I tried also with 0xF021, 1, but same. I am able to ping TwinCAT computer, running it online, get connection but not read. I did also mapping, check configuration, activate configuration and get TwinCAT running.

Please below is the screenshot. 
[image]
Thank you so much for any help

Reyan

xift, 2016-08-01:
Hey,
"connection closed by remote" indicates that you have no route to your ADS client on your TwinCAT machine. If your ADS client runs on 192.168.10.1 add a TCP route to that ip with the Ams net id of 192.168.10.1.1.1.

It took me multiple days to figure out that that was the problem all along :-)
Maybe we should add it to the readme.

Good luck!
Michael

reyanvaldes, 2016-08-01:
Hi Michael
Thank you so much for quickly replying and sharing, it is a great help!. I put route as static on TwinCAT project under Route Settings, and message "connection closed by remote" is gone, but still getting read failed, now with different error code: 1799. To be honest, no idea. I tried to read different things but no success, at least it is a little better than before.
[image]

Thank you so much, appreciate any help
Reyan

xift, 2016-08-01:
Glad I could help.

Never seen 1799 before... It means "device not ready". Try to run the readState example first to see which state the device is in.

Good luck
Michael

reyanvaldes, 2016-08-01:
Michael
Thanks again, It is shown state: 6, dev state:1. I don't 't know what does it means. 

Good news!!!, I did new fresh downloading, and now it is able to read good values!!! 

I was jumping here on my seat!!. 

Thanks a lot, it was a great help!!!!

soberschmidt, 2016-08-02:
Please check the readme. The way to add route is described. Furthermore you will find the documentation to the ADS return codes and TwinCAT state enumerations.

closed 2016-08-02 by soberschmidt

xift, 2016-08-02:
I know that it is described in the readme file. However, if multiple people run into the same problem it might be worth to highlight that information.

Regards 
Michael

reyanvaldes, 2016-08-02:
Appreciate a lot the great help from you, and sharing nice library work. As a new user, I have only a couple of comments, that maybe could help other people like me (beginner in Beckhoff):

1- The way I read first from readme I misunderstand because I was thinking was mentioned about the example cpp itself, which has AdsAddRoute and I though with it was enough. For some reason we have to have it on the cpp and also explicitly set on TwinCAT project with the client IP, which is in my mind still is a little confusing, typically the comm server handle the clients request doesn't need to define at server itself which clients will be connected. It could be nice, if possible, to have option to add this from the client program also without touching the server it self.

2- Some link referenced on AdsLib.h works but others I couldn't opened correctly, for example in AdsSyncReadDevice, the adsState link info works, but the return code doesn't. It is a little difficult for a person not familiar with Bechkoff try to find the meaning of these codes. Maybe could be helpful to have a functions that return text message for: Ads State, Ade device status, and return codes. 

Thank you so much for great work!!! and shared with community

Note: By the way I created kind of very simple ads_map (mimic what it is on S7 in Snap7 library) to handle different data types with read/write operation on buffer, not sure if like to take a look. The idea is to handle different data types in/from buffer, and make it easy for parsing in client program. I believe could help to handle easier the reading or writing data.

Bests
Reyan

techaitr, 2021-02-11:
@reyanvaldes Can you explain that how is Adslib define in .pro file?