Beckhoff/ADS #34: Alarm&Events through ADS

rockonedege, 2016-08-02:
The [documentation](http://infosys.beckhoff.com/english.php?content=../content/1033/tceventlogger/html/introduction.htm#formatter&id=20116515212232807725) indicates it's possible to access TcEventLogger remotely with ADS.  However the [API details](http://infosys.beckhoff.com/english.php?content=../content/1033/tceventlogger/html/Connectivity/ADS/Interfaces/ITcEventLogAdsProxy.htm&id=) seem to say only Windows CE accepts such remoting.  

- TcEventLogger seems to be a separate COM server, is it accessible from the same TCP ports this project uses.
-  Is the support of fetching alarms and Events from TcEventLogger  within the scope of this project if the answer to the above question is yes?

soberschmidt, 2016-08-03:
The Eventlogger is also available for BigWindows and can be accessed via ADS or DCOM.

closed 2017-01-05 by pbruenn

Mebus, 2025-01-24:
Hallo,

could you please give me some hints or examples on how to access the events logged by the event logger through the C++ ADS library?

Thanks!

Mebus

hmersch-bk, 2025-01-28:
For the Tc3Eventlogger (https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_eventlogger/index.html&id=8504177607767980219) we are working on a C++ API compareable to the AdsProxy existing for C#

Mebus, 2025-01-28:
Hey,

that's good to hear.

I fiddled around with the C# version a little bit and noticed that it is not Linux compatible :-(

Mebus