Beckhoff/ADS #49: Multiple connections to a single TwinCAT computer.

eatli, 2017-09-13:
Hi, I'm having trouble connecting multiple instances of the same AdsLib application to a single TwinCat computer. 

Program instance 1. 
AdsAddRoute
AdsPortOpenEx
while(1) 
{
  read/write some data on interval. 
}

This works fine but if i start another instance of the same program and try to connect to the same TwinCat computer, program 1 fails and I get ClientSyncTimeout(1861) as soon as the program instance 2 hits AdsAddRoute. 

Is this just not an option in the implementation of the ADS, or further more how would you achieve the same functionality in case I have overlooked something? 

Another thing I've noticed is that reading the same variable with AdsSyncReadWriteReqEx2  doesn't take constant time in fact it varies as you can see on the image below(from 0.5-2,5 ms). Is this just something to be expected? 

[image]

Any help/pointers would be appreciated. 

Regards Emil.

pbruenn, 2017-09-18:
Hi Emil,
you see two limitations of AdsLib/ADS.
1. TwinCAT AMS Router doesn't allow multiple TCP connections from the same host. So when you use two AdsLib instances on the same host to connect to the same TwinCAT router, you will see that TwinCAT will close the first TCP connection and only respond to the newest.
2. As ADS is transmitted over a TCP connection, there is no real time guarantee.

The first limitation could be fixed by implementing a real router within AdsLib. Or you can try to use different IP addresses for your applications.

closed 2017-09-18 by eatli

juwalter, 2022-01-28:
While it seems indeed not possible to connect to multiple instances using the library on Linux, it is possible to run multiple instances of your application, and each application connecting to the same or different instance of an ADS Server/Beckhoff PLC.

In order to do so, it is necessary to 
a) use multiple IP addresses on Linux - either on different network interfaces or on the same network interface
b) register multiple "routers" on the remote PLC: one entry in the Beckhoff router for each IP address on Linux
c) use some `iptables` magic on Linux to make traffic appear to come from different IP addresses from the perspective of the PLC

Overview:

i) general setup: 1x PLC with IP address `192.168.1.10/24`, 1x Linux machine with IP addresses `192.168.1.20/24` and `192.168.1.21/24` on same interface
ii) additional "virtual" IP address `192.168.1.11/24` for PLC; this address will not be configured on the PLC, rather it is only used to be able to apply `iptables` rules in the Linux machine
iii) configure application A to use destination address `192.168.1.10/24` for PLC connection
iv) configure application B to use destination address `192.168.1.11/24` for PLC connection
v) configure Linux to rewrite virtual destination address `192.168.1.11/24` as `192.168.1.10/24` (so traffic actually goes to the PLC at address `192.168.1.10/24`)
vi) configure Linux to set src field of IP packets to second IP address `192.168.1.21/24` when destination is address `192.168.1.11/24`

```
sudo iptables -t nat -A OUTPUT -d 192.168.1.11 -j DNAT --to-destination 192.168.1.10 # see "v)" above
sudo ip address add 192.168.1.21/24 dev enp1s0 # add secondary IP address to Linux network interface, see "i)" above
sudo ip route add 192.168.1.11/32 dev enp1s0 src 192.168.1.21 # see "v)" above
```

hat tip to user "A.B" at https://serverfault.com/questions/1090892/linux-make-traffic-from-same-host-appear-to-come-from-different-ip-addresses/1090918#1090918