Beckhoff/ADS #53: Multiple Request/Response

pillh, 2017-12-04:
Support for multiple requests (read/write) is missing. Implementing these would allow fast access to several variables.

pbruenn, 2017-12-05:
Do you mean this https://github.com/Beckhoff/ADS/issues/3 ?

pillh, 2017-12-05:
It is close, but not right there. I need to get a set of variables from the PLC and I do have their names, but not their location. So at the moment I cyclically get the handle, read the variable and release the handle for each variable. That produces a lot of traffic.
You do have an example for getting several handles at once, but the functions supplied for reading/writing variables (AdsSyncReadReqEx/AdsSyncWriteReqEx) using the handle do not seem to support multiple variables. Or do I miss something?

pbruenn, 2017-12-05:
Did you check [ADS-DLL Sample17.zip](https://infosys.beckhoff.de/content/1031/tc3_adssamples_c/html/tcadsdll_api_cpp_sampleintro.htm?id=8736397773043145650)?

Note: I am not an ADS expert, but as far as I can see our sample is not 100% correct. In my opinion it would read the float, two times, because the second:
`pAdsSymbolEntry       = (PAdsSymbolEntry)pInfoExFloat` 
should be
`pAdsSymbolEntry       = (PAdsSymbolEntry)pInfoExDword`

rhythmchicago, 2017-12-05:
Why do you get the handle every time? My practice is to get the handle, store it, use it while I’m running the program, then release when shutting down the program. A handle remains valid unless the PLC program is changed, and you can register a signal to detect that has happened and then refresh the handles.

Also, I tend not to scan cyclically; I register notifications on the handle and use the On Data Change flag. A callback will be fired anytime. If you must have cyclical updates, you can register the notification to happen on a cyclic timer instead.

Thomas Bitsky Jr
V.P. Product Development
Automated Design Corporation
1404 Joliet Road, Suite D | Romeoville, IL 60446
P: (630) 783-1150 | F: (630) 783-1159 | M: +1 6306326679
automateddesign.com<http://www.automateddesign.com>

[Facebook]<https://www.facebook.com/automateddesigncorp/>  [Facebook] <https://www.instagram.com/automateddesign/>   [Facebook] <https://twitter.com/adcsportslogic>   [Facebook] <https://www.youtube.com/user/ADCSportsLogic>

Disclaimer

This message contains confidential information and is intended only for the individual(s) addressed in the message. If you are not the named addressee, you should not disseminate, distribute, or copy this e-mail. If you are not the intended recipient, you are notified that disclosing, distributing, or copying this e-mail is strictly prohibited.

pillh, 2017-12-05:
I do know that there is room for optimization using the functionality already available. That was NOT the initial question.
The reason why i do not use notifications is that I already have bad experience with other protocols (some tags do not get sent anymore).
During commissioning the Location of the variables will Change a lot, thats why i initially intended to get the handles every single time.
At the Moment I'm thinking about reading the INFO (ADSIGRP_SYM_INFOBYNAMEEX 0xF009) and using the stored info for the grouped request.
The initial issue however was about the missing support of the LINUX ADS stack and it's interfaces for grouped variable access.
So far it is still missing, even the structure definition used in sample 17 (AdsSymbolEntry) is not available in the LINUX headers.
If a sample for a multiple variable access using the  given Interfaces would be added, I'd be satisfied.
So far I'll just try to puzzle it out and add it myself.

pbruenn, 2023-01-12:
struct AdsSymbolEntry was merged with commit 5c7c8d9e66920ef37d0f4384fb841e6f7b406d4d

closed 2023-01-12 by pbruenn