Beckhoff/ADS #54: Hardcoded frame size

hvdbrand, 2017-12-06:
https://github.com/Beckhoff/ADS/blob/7df1d60f8bf088a2be718323d27aa2156f7d7b0f/AdsLib/AmsConnection.cpp#L27

Dear sir/madam,

We are using this library to interface with a Beckhoff PLC.
We would actually like to receive ADS results that are larger than 4096.

Therefore we were wondering why the frame size is hard coded to 4096?
Is it possible to extend it?

If it cannot be extended, what would be the preferred way to obtain an object from the PLC that is larger than 4096 bytes?

Thanks in advance for your response.

Best wishes,
Hugo van den Brand

pbruenn, 2017-12-06:
Hello Mr. van den Brand,
there is no special meaning behind this 4096 bytes. When I started this library it was just enough and until today nobody complained.
Next week I will try to find some time to make the interface more dynamic. Until then could you try to use a different hard coded size?
Please let me know what size you use and if it was working, so I have a hint for my own test cases.

Best regards,
Patrick

hvdbrand, 2017-12-06:
Hi Patrick,

Thank you for your very fast response.
I have changed the number to 30000 bytes (because I needed to read 13978 bytes).
This worked well.

Best wishes,
Hugo

pbruenn, 2018-01-03:
Hi Hugo,
can you confirm that commit aa2ba7b fixes your issue?
if it doesn't please reopen this issue.
Thanks,
Patrick

closed 2018-01-03 by pbruenn

hvdbrand, 2018-01-03:
Based on the code changes this seems to work.
I will check it if I find some time.

commit aa2ba7bf (2018-01-03): AdsLib: add support for AmsResponses > 4096 bytes

This fixes #54 "Hardcoded frame size"