Beckhoff/ADS #57: Unclear README instruction: How do we download the example to the target?

ceremcem, 2018-03-10:
Maybe a total newbie question, but to get up and running quickly, I'm trying to load the example PLC code to the simulator of TwinCAT, but [the following instruction](https://github.com/Beckhoff/ADS/blob/7df1d60f8bf088a2be718323d27aa2156f7d7b0f/README#L28) has no meaning to me: 

>Download your PLC project to your target e.g. "PLC-TestProject" of our GitHub repository.

What does that mean? `.sln` files are C# project files AFAIK, how can I load a C# project to the PLC?

pbruenn, 2018-03-12:
Indeed, you need to compile the PLC project and then upload it to your PLC.
This README cannot cover all aspects of ADS and TwinCAT so I like to refer you to our Infosys:
https://infosys.beckhoff.com/content/1033/tc3_plc_intro/2525041803.html?id=7770895129335440874

Two more common problems:
1. You cannot use this OpenSource lib to connect to a TwinCAT server on localhost.
2. Make sure you setup your [ADS/AMS routes](https://github.com/Beckhoff/ADS/issues/14#issuecomment-362547370) correctly

pbruenn, 2018-03-12:
Yes, PLC-TestProject is a TwinCAT 3 project.
If you were able to connect to your TwinCAT server with node-ads. It should be straightforward to adjust example/example.cpp to read exactly the same values.

ceremcem, 2018-03-15:
I'm currently not capable of editing the `example/example.cpp` but this issue can be considered as "answered". Thank you.

closed 2018-03-15 by ceremcem