Beckhoff/ADS #60: Error: Connect TCP socket failed with: 110 ?

lianhub, 2018-03-17:
I am wondering if I understood the programming model correctly.
Here is the IP info:
* My Windows-PC with TwinCAT-3: 192.168.2.41
* My Linux-target:                           192.168.2.119
Here is what I did:
* On Linux target:
* --- git clone https://github.com/Beckhoff/ADS.git
* --- cd ADS
* --- make
* --- cd example
* --- modify example.cpp
  ```
 static const AmsNetId remoteNetId { 192, 168, 0, 231, 1, 1 };
   static const char remoteIpV4[] = "ads-server";
```
* --- make
* --- make test

pbruenn, 2018-03-19:
On the first look, everything seems correct.
1. can you ping 192.168.2.41 from your Linux machine?
2. Does your Windows Firewall accept connections on ADS Port (48898)

lianhub, 2018-03-19:
pbruenn:

Thanks for your information!
* Yes, I can ping 192.168.2.41 from my Linux machine.
* How do I check if my firewall accepts connections on ADS port (48898)?
* --- I have used TwinCAT-3 to communicate with Beckhoff CX5020 and control AX5206.
* --- so I guess that ADS port is already allowed, right?

Thanks!

pbruenn, 2018-03-20:
If you have run TwinCAT3 from a different machine than 192.168.2.41, then yes that port should be open.
If you just run it on localhost (192.168.2.41), than your firewall wouldn't intervene.
In case you didn't run TwinCAT from a different machine try this:
run `telnet 192.168.2.41 48898` on your Linux machine depending on the result you see:
- connection refused means that nothing is running on that port
- accepted means that something is running on that port
- timeout means that a firewall is blocking access

You find more details here:
https://stackoverflow.com/questions/273159/how-to-determine-if-a-port-is-open-on-a-windows-server

lianhub, 2018-03-20:
OK, I run `telnet 192.168.2.41 48898`, and it failed with 'timeout'!
Now I got further question:
* My Windows-PC has two ports
* Port-1: 192.168.2.41 
* ----- used for regular internet, 
* ----- used for connection to ADS@Linux via router (what I am testing now)
* Port-2: used for TwinCAT-3 
* ----- connected to CX5020 then AX5206
* ----- or connected to AX5206 directly

My question is:
* doesn't CX5020 has ADS client also? so shouldn't ADS port be allowed already?

Thanks,
Jerry

pbruenn, 2018-03-20:
Hmm, now I am confused.
Where is your TwinCAT EtherCAT master running? On Windows-PC and CX5020 or only on the CX5020?

When you connect from Linux to 192.168.2.41:48898 the firewall on 192.168.2.41 is blocking the _incoming_ port 48898
But when you connect from 192.168.2.41 to CX5020:48898 then the firewall on 192.168.2.41 don't necessarily has a rule to block outgoing connections to 48898 on some remote machine.

In fact I am pretty sure Windows standard firewall will allow everything outgoing, or just ask you to open the outgoing port.

If you really want to connect from Linux to 192.168.2.41, make sure you add some exception to your firewall. But I suspect you want to connect from Linux to your CX5020, don't you? In that case with that network configuration you would have to enable IP routing on your windows machine@192.168.2.41 (Don't ask me how to do this, I don't know ;-)

lianhub, 2018-03-20:
OK, I guess I misunderstood the programming model/function of this ADS repository.
**Here is what I want:**
* use TwinCAT-3 on Windows to connect (via ADS) EtherCAT master on Linux!

**Here is what I understood:**
* There are EtherCAT masters on both Windows-PC and CX5020, right?
* There are ADS server/client  on both Windows-PC and CX5020, right? 
* When TwinCAT-3 connects to CX5020
--- TwinCAT-3  ADS -----  talk to --------CX5020 ADS-------- talk to -------CX5020 EtherCAT master, right?

**Here is what I don't understand:**
* When TwinCAT-3 connects to CX5020, who is server, who is client:
----- TwinCAT-3 ADS: server, client?
----- CX5020 ADS: server, client?

Thanks!

pbruenn, 2018-03-20:
Okay, did I understand correctly:
You want to run an ADS server on Linux and then connect to that server with TwinCAT3, just like you use TiwnCAT 3 Windows-PC to connect to CX5020(with TwinCAT)?
That's not possible with this implementation . Right now only client mode is implemented. And even that is limited as there is no real ADS router.
For your other questions I assume you are running TwinCAT on CX5020, too.
Then yes, there are EtherCAT masters on both your Windows PC and CX5020. In case you don't have an active configuration on your windows PC, the EtherCAT master on it will be disabled.
And yes, ADS server/client is running on both Windows-PC and CX5020(with TwinCAT)
Then yes, TwinCAT on your Windows PC will talk to ADS router(server) on your CX5020 and the router will forward to the EtherCAT master
In that case Windows PC will be "client".

Not sure if that was helpful...

lianhub, 2018-03-20:
Yes:
`You want to run an ADS server on Linux and then connect to that server with TwinCAT3, just like you use TiwnCAT 3 Windows-PC to connect to CX5020(with TwinCAT)?`

Yes: 
`For your other questions I assume you are running TwinCAT on CX5020, too.`

Thanks pbruenn:
* You answered all my questions, so that I understand the function/limitation of this ADS repository.

One last question:
* Why do we need an ADS client on Linux? we can always use TwinCAT-3 as ADS client, right?
* What could be the possible use of this ADS client?

Thanks again!

pbruenn, 2018-03-20:
Oh! Okay, I never expected that constellation.
As far as I know the ADS protocol is implemented only by Beckhoff. So for us the "natural" use case is a customer who operates a plant with machines from different vendors. Some of these vendors use TwinCAT to control their machine. But in case the plant operator runs Linux on his management PC, he needs a way to communicate with TwinCAT. That's the point where this AdsLib comes in. You can use it on Linux to build a client application to query and set values on your TwinCAT machines.

If you want to take this library and advance it for server side usage, feel free to do so. And of course it would be great if you share your work , so I could try to merge it back here.

Regards,
Patrick

lianhub, 2018-03-20:
Thanks Patrick very much for your insightful explanation!
* Yes, I will try to extend this library to server side.
* I will study the code and raise some questions later.

For now, we can close this issue!  Later, I will raise questions by open new issues.

pbruenn, 2018-03-20:
I am looking forward to it!

closed 2018-03-20 by pbruenn