Beckhoff/ADS #61: A bug?

lianhub, 2018-03-20:
I think I may find a bug:
```
* I modified scokets.cpp
uint32_t TcpSocket::Connect() const
{
    const uint32_t addr = ntohl(m_SockAddress.sin_addr.s_addr);
	printf("The IP address is: %s : %d : %d\n", inet_ntoa(m_SockAddress.sin_addr), m_SockAddress.sin_port, m_SockAddress.sin_family);

    if (::connect(m_Socket, reinterpret_cast<const sockaddr*>(&m_SockAddress), sizeof(m_SockAddress))) {
        LOG_ERROR("Connect TCP socket failed with: " << WSAGetLastError());
        return 0;
    }
```
* then

Any can confirm my judgment?

Thanks,
Jerry

pbruenn, 2018-03-20:
Unfortunately that's no bug. sin_port is in "network byteorder/big endian" so you would have to convert it back into host byteorder with ntohs() and will see it is correct.
You could use Wireshark[[1](https://www.wireshark.org)] to see the packet on wire (in network byte order, but wireshark will decode correctly).

[1] https://www.wireshark.org

lianhub, 2018-03-20:
OK, I see the logic!
Thanks Pbruenn, so this issue can be closed now!

closed 2018-03-20 by pbruenn