Beckhoff/ADS #62: How to test this ADS client?       (two problems)

lianhub, 2018-03-20:
Patrick, I got **two problems** for testing this ADS client.

**First problem:**
* When I open "PLC-TestProject/NotificationTest.sln"
* I saw I/O components: **EK1200-EL6900-EL1904-EL6910-EL6930**   (why three EL69xx?)
* Do we need to connect all these components for testing?
* Can the PLC-TestProject be simpler? (I have EK1100, EL1004, EL2004, and EL6900, EL1904)

**Second problem:**
* I build "example" in Linux successfully!
* But when I tried to build them in VisualStudio-2015 (with Visual C++ 2015 installed)
* I got errors:
```
Error	LNK1104	cannot open file 'D:\myGithub\ADS\Debug\/AdsLibOOId.lib'
Error	C1083	Cannot open include file: 'fructose/fructose.h': No such file or directory
```

Please help me to solve the problems!
Thanks,

Jerry

pbruenn, 2018-03-21:
Hi Jerry,
no, you don't need any hardware at all. You can just download the PLC project and ignore all the hardware configuration. 
`Error	LNK1104	cannot open file 'D:\myGithub\ADS\Debug\/AdsLibOOId.lib'`
Was a bug and should be fixed by eb9a25fd6362a427f435f5311bfd48f625ef7449

`Error	C1083	Cannot open include file: 'fructose/fructose.h': No such file or director`
Shouldn't occur if you only compile "example" and it's dependency "AdsLib".
Fructose is just included by "AdsLibTest" and "AdsLibTestRef" and they indeed need another fix  59b42806baa3b6793223ff68a7e755214de09c91
You can install the Fructose C++ Unittest framework by running ` ./tools/10_get_fructose.sh` in an UNIX environment like Cygwin, or just read the script and download + extract fructose manually.

Another side note: You stumbled up on AdsLibOOI. Which is an attempt to provide a more C++ like API to ADS. You find the latest work on branch [dev-ooi-v2](https://github.com/Beckhoff/ADS/commits/dev-ooi-v2). It might be a better start to implement a full router and ADS server than the legacy C API.
Right now, OOI is implemented on top of C API. If you are scared by that because of performance (and code style), I can absolutely understand, but be assured that the long term idea is to invert that and convert the whole AdsLib to C++ style with an C API on top to support legacy applications.

lianhub, 2018-03-21:
Thanks Patrick;
* Now I removed the dependency on "AdsLibOOI", and compiled example successfully.
* I will skip "AdsLibtest" and "AdsLibTestRef" for the time being.
* --- but can you describe briefly **what are the function of these two projects?** 
* I will skip "OOI" topic for the time being.
* Now I got a license problem when trying to download PLC-TestProject:
[image]

Any hint? 

Jerry

pbruenn, 2018-03-21:
I use "AdsLibtest" and "AdsLibTestRef" as regression test for AdsLib. Difference between them is AdsLibTestRef is linked against TcAdsDll.dll (our Windows only reference implementation)

I don't know a lot about TwinCAT and licensing, but from your screenshot I would expect something is wrong with your clock or date. For testing you should be able to just use a 1 week trial license

lianhub, 2018-03-21:
I already regenerated 7 Days Trial license upon request.
* By the way, I  just flashed TwinCAT-3 to CX5020 (which installed TwinCAT-2 originally), for the purpose of testing.

Weird, it never happens before.
I guess I should raise this issue to Beckhoff Support team.

lianhub, 2018-03-21:
Hi, Patrick:

All my problems fixed. Testing "example" runs smoothly!

I think more details should be added to README file, as there are some untold tricks to be done, depending on the setup of testing platform. In my case:
* you should add: **AdsSetLocalAddress({192, 168, 2, 41, 1, 1});**
* you should close TwinCAT-3 after downloading PLC-TestProject, otherwise, running "example.exe" will fail.

Now I will move forward: study code, maybe hop on OOI version, surely more questions later.
Thank again!

commit 6ebe7a2f (2018-03-22): README: explicitly mention AdsSetLocalAddress() might be useful

fixes #62
closed 2018-03-22 by pbruenn