Beckhoff/ADS #65: Connection Failed

TamakiAutomation, 2018-04-29:
I'm trying to connect CX2020 to PC by pyads.

IP address is followings, but EAP3 port is connected to my pc.
And connection iteselc is established.

I tried on other NetId and IP address combination. But i couldn't find answer.
And Port is set by twincat3, 27908.(iI tried also pyads.PORT_SPS1)

As pyads code,
`pyads.Connection('5.53.192.206.4.1,27908,169.254.193.18)`

location | AmsNetId | Address
Desktop | 192.168.0.15.1.1 | 169.254.170.72
EAP3 | 5.53.192.206.4.1 | 169.254.193.18
EAP2 | 5.53.192.206.3.1 | 169.254.37.249
EAP1 | 5.53.192.206.2.1 |  
Target NetId | 5.53.192.206.1.1 |  
Local NetId |   | 192.168.0.15.1.1

How should I make the configration?

pbruenn, 2018-04-30:
From what I understand 851 should be the port your are looking for.
https://github.com/Beckhoff/ADS/blob/c1b039f2e507ca17abdddcacd7be24d5853ed9e9/AdsLib/AdsDef.h#L37

If that doesn't help please ask at pyads https://github.com/stlehmann/pyads. They should know better what kind of port pyads.Connection() is expecting

TamakiAutomation, 2018-04-30:
Thank you for feedback, but it didnt work. I have to chagne other part of config also.
I'll follow your recommendation pyads post.

closed 2018-04-30 by TamakiAutomation