Beckhoff/ADS #71: Error--Add device notification failed with: 1861

xuyuan25, 2018-07-17:
Hi support,
  i want to use your code connect beckhoff PLC(twincat2,winCE) on linux without twincat.i edit example.cpp like this:
[image]
  there are some error like this:
[image]
 thanks a lot for your recommendation!
Yuan

pbruenn, 2018-07-17:
On your TwinCAT2 system, did you add a route for your Linux system with AmsNetId "5.21.95.61.1.1"?

xuyuan25, 2018-07-17:
thank you for your answer
now I add a route with AmsNetID"5.21.95.61.1.1" on TwinCAT2 system,and still has error:
 
[image]
thanks a lot for your recommendation!
Yuan

xuyuan25, 2018-07-19:
hi,Patrick
i have solve this error by change the port,so i can read something.
thanks.

Now,I'm try to write something into beckhoff PLC.I find some example from https://infosys.beckhoff.com/english.php?content=../content/1033/tcsample_vc/html/tcadsdll_api_cpp_sample17.htm&id=20654114132734386906 so I edit code and there are no error, but beckhoff cannot received what i write.
if you have suggestion or example，you can tell me.
 thanks a lot for your recommendation.
yuan

pbruenn, 2018-07-19:
If I understand correctly `AdsSyncReadReqEx2()` is working, now.
I would try these steps next:
1.Test if writing to the PLC is working with [`AdsSyncWriteReqEx2`](https://github.com/Beckhoff/ADS/blob/master/AdsLib/AdsLib.h#L145)
2. Test if the sum command is working, by using it to only read.

closed 2018-11-13 by pbruenn