Beckhoff/ADS #72: Multiprocess use?

stobor827, 2018-07-18:
Should this library be able to be used in two separate processes to connect to the same ADS server?
I get a 'connection closed by remote' error on the ADS_TCP_SERVER_PORT (0xBF02) when a second application starts on the same PC. Is there a routing trick or some other way to work around this?

pbruenn, 2018-07-19:
Short answer: No.
If you use TwinCAT on both PCs and have two applications on on PC. All ADS requests of both applications are multiplexed by the TwinCAT ADS router to use the same TCP/IP connection.
That's why the TwinCAT router on the target machine doesn't expect a second TCP/IP connection from the same host.
If I would have to use this library for two applications I would see two options:
1. advance this library to be split into a central router service and a library to connect to this local router, like it's done in TwinCAT
2. Use interprocess communication to let both applications talk to a third one, which then uses this library to send the AdsRequests

Regards,
Patrick

stobor827, 2018-07-19:
Thanks Patrick.
I have TwinCat installed on the client PC and have a route configured through its router.  Is it possible to leverage that to forward ADS Requests for me instead of implementing a new router as proposed in 1?  I feel like users of the .NET libs and maybe the TcADSDll.dll can do this?  Sorry if this is a dumb q, I don't fully understand the entire TwinCat ecosystem.

I'd be willing to take a stab at necessary changes for that, but I don't need this feature enough for the work involved with 1 or 2.

pbruenn, 2018-07-20:
If you have TwinCAT on the client, you can simply use TcAdsDll.dll instead of AdsLib (my name for this open source library).
AdsLib was designed to be compatible to TcAdsDll.dll. Differences are:
- When there is a function with multiple variants such as AdsSyncReadReq/Ex/Ex2 only the latest variant is implemented by AdsLib
- AdsLib API is more const correct

So to have an application to link against AdsLib or TcAdsDll.dll:
- use only the functions available within AdsLib
- don't provide pointers to const data to AdsLib

Take a look into  [AdsLibTest](https://github.com/Beckhoff/ADS/blob/master/AdsLibTest) (uses AdsLib) and [AdsLibTestRef](https://github.com/Beckhoff/ADS/blob/master/AdsLibTestRef) (uses TcAdsDll.dll).

When you compare the main.cpp of both projects you will see the little differences I mean. AdsLibTestRef should link against AdsLib without any change.

closed 2018-11-13 by pbruenn