Beckhoff/ADS #76: Provide compiled binary

gimmi, 2018-10-29:
Hi, 

any plan to provide pre-built binaries for this lib?

pbruenn, 2018-10-30:
Actually no. What would be the use case for pre compiled binaries?

gimmi, 2018-10-30:
I am evaluating the use of this lib to be wrapped for use in .NET, and I'm having trouble setting up all the environment needed to compile it. Having an already compiled version would be useful.

pbruenn, 2018-10-30:
.NET sounds like you are on Windows. Then you could just use the official prepackaged libraries from Beckhoff:
https://infosys.beckhoff.de/content/1031/tc3_adsdll2/html/tcadsdll_intro.htm
https://infosys.beckhoff.de/content/1031/tc3_adsnetref/4017095947.html

If you are familiar with Docker here is a Dockerfile for a Ubuntu docker container, which is able to cross compile for windows:
```
from ubuntu:16.04

RUN apt-get update && apt-get install -y \
autoconf automake autopoint \
bash bc bison build-essential bzip2 \
ca-certificates \
clang \
cmake \
flex \
gettext git g++-multilib gperf \
intltool \
libc6-dev-i386 libffi-dev libgdk-pixbuf2.0-dev libtool libtool-bin libltdl-dev libssl-dev libxml-parser-perl \
nmap \
openssl \
p7zip-full \
patch perl pkg-config python \
ruby \
scons sed \
unzip \
wget \
xz-utils

RUN git clone https://github.com/mxe/mxe.git
RUN cd mxe && make -j`nproc` MXE_TARGETS=i686-w64-mingw32.static.posix gc
```

gimmi, 2018-10-30:
Thanks a lot for the Docker hint, much appreciated.

Will try with it.

closed 2019-04-26 by pbruenn