Beckhoff/ADS #77: Add device notification failed with: 1797

tamberg, 2018-11-01:
Hi,

trying to run the C++ example (on macOS) to connect to your "PLC-TestProject" on TwinCAT 3 running on a Windows 10 Laptop.

Connecting seems to work (ping and telnet is fine, output below says "Connected") but then there's 

> Add device notification failed with: 1797

Any ideas?

`$ make test
./example.bin
2018-11-01T12:48:48+0100 Info: Connected to 192.168.2.77
2018-11-01T12:48:48+0100 Warning: Frame to long: 40<8
Add device notification failed with: 1797
notificationByNameExample():
2018-11-01T12:48:48+0100 Warning: Frame to long: 30<12
Create handle for 'MAIN.byByte[4]' failed with: 0x705
2018-11-01T12:48:48+0100 Warning: Frame to long: 40<8
Add device notification failed with: 1797
readExample():
ADS read failed with: 6
readByNameExample():
2018-11-01T12:48:48+0100 Warning: Frame to long: 30<12
Create handle for 'MAIN.byByte[4]' failed with: 0x705
ADS read failed with: 6
ADS read failed with: 6
2018-11-01T12:48:48+0100 Info: connection closed by remote`

According to [ADS Error Codes](https://infosys.beckhoff.com/english.php?content=../content/1033/devicemanager/36028797393240971.html&id=) *1797* means
> ADSERR_DEVICE_INVALIDSIZE | parameter size not correct

pbruenn, 2018-11-02:
Weird. When I get these errors I forgot to close my TwinCAT XAE session.
But you said you are using macOS. Do you run a virtual Windows with XAE on the macOS device, which might share the IP address (because of NAT)?
In that case set the remote target to local or close the TwinCAT XAE and try again.
If that scenario doesn't match your environment or the problem persists, could you make a Wireshark capture between your macOS and Windows 10 Laptop?

If you run TwinCAT XAE on your Windows 10 Laptop connected to localhost, please try to close that, too.
I have no logical explanation why that should disturb your connection to the macOS, but please just try it ;-).
regards, Patrick

tamberg, 2018-11-02:
Hi, TwinCAT runs on a Win10 Laptop (which I think is the ADS Server), and the C++ example in this repo runs on my macOS Laptop.

> try to close that

Ok, thanks a lot.

soulzhao, 2018-12-04:
> Hi,
> 
> trying to run the C++ example (on macOS) to connect to your "PLC-TestProject" on TwinCAT 3 running on a Windows 10 Laptop.
> 
> Connecting seems to work (ping and telnet is fine, output below says "Connected") but then there's
> 
> > Add device notification failed with: 1797
> 
> Any ideas?
> 
> ```
> $ make test
> ./example.bin
> 2018-11-01T12:48:48+0100 Info: Connected to 192.168.2.77
> 2018-11-01T12:48:48+0100 Warning: Frame to long: 40<8
> Add device notification failed with: 1797
> notificationByNameExample():
> 2018-11-01T12:48:48+0100 Warning: Frame to long: 30<12
> Create handle for 'MAIN.byByte[4]' failed with: 0x705
> 2018-11-01T12:48:48+0100 Warning: Frame to long: 40<8
> Add device notification failed with: 1797
> readExample():
> ADS read failed with: 6
> readByNameExample():
> 2018-11-01T12:48:48+0100 Warning: Frame to long: 30<12
> Create handle for 'MAIN.byByte[4]' failed with: 0x705
> ADS read failed with: 6
> ADS read failed with: 6
> 2018-11-01T12:48:48+0100 Info: connection closed by remote
> ```
> According to [ADS Error Codes](https://infosys.beckhoff.com/english.php?content=../content/1033/devicemanager/36028797393240971.html&id=) _1797_ means
> 
> > ADSERR_DEVICE_INVALIDSIZE | parameter size not correct
> 
> (I'm new to TwinCAT, could well be a basic issue on that side.)

Maybe the same issue:
https://github.com/Beckhoff/ADS/issues/80#issue-386660994

soulzhao, 2018-12-04:
Maybe the same issue:
https://github.com/Beckhoff/ADS/issues/80#issue-386660994

closed 2019-04-26 by pbruenn

MarkRiddoch, 2019-10-29:
Did you ever find a solution to this? I am trying something similar, but this time with the example code running on a Linux machine and the Twincat running in a Windows 7 VM on my local machine. I get identical output to yours.

> ADS/example$ ./example.bin 
> 2019-10-29T10:26:56+0000 Info: Connected to 192.168.0.128
> 2019-10-29T10:26:56+0000 Warning: Frame to long: 40<8
> Add device notification failed with: 1797
> notificationByNameExample():
> 2019-10-29T10:26:56+0000 Warning: Frame to long: 30<12
> Create handle for 'MAIN.byByte[4]' failed with: 1797
> 2019-10-29T10:26:56+0000 Warning: Frame to long: 40<8
> Add device notification failed with: 1797
> readExample():
> ADS read failed with: 6
> readByNameExample():
> 2019-10-29T10:26:56+0000 Warning: Frame to long: 30<12
> Create handle for 'MAIN.byByte[4]' failed with: 1797
> Unable to determine symbol size, reading ADS symbol information failed with: 6
> 2019-10-29T10:26:56+0000 Info: connection closed by remote

I've closed everything on the windows side (the Manager and Control were all I had opened).

soulzhao, 2019-11-28:
Hi  @MarkRiddoch , check my post, maybe it will help::
https://github.com/Beckhoff/ADS/issues/80#issue-386660994