Beckhoff/ADS #79: Adding ADS route failed

ZhousuLee, 2018-12-01:
I am trying to run the C++ example (on Ubuntu 16.04) to connect to your "PLC-TestProject" on TwinCAT 3 running on a personal computer with windows7 system .
     I set IPV4 address of windows 7 to 192.168.0.231 and set AMSNetId of Twincat3 to 192.169.0.231.1.1
     I set IPV4 address of Ubuntu 16.06 to 192.168.0.1 
     When I run the c++ example (on Ubuntu 16.04)  after running "PLC-TestProject" on TwinCAT 3 ，Ubuntu always show me that “Adding ADS route failed”.
    I have tried ADS communication between two personnal computer both with windows system, and it works well. I could add route  manually by click " Router/Edit routes" because both of  computers have Twincat3. So I am not sure whether i need do this when one of computers is Ubuntu system.
    I am new to ADS so it maybe very easy.
    Thanks a lot.

pbruenn, 2018-12-03:
I guess this was just a typo? "192.168.0.231 and set AMSNetId of Twincat3 to 192.169.0.231.1.1" and you meant "192.168.0.231.1.1"

Then the remoteNetId in the example would already be correct:
https://github.com/Beckhoff/ADS/blob/781d7668502de67afbaae07130088da5c6f61e45/example/example.cpp#L180

But did you adjust the remoteIpV4, too?
https://github.com/Beckhoff/ADS/blob/781d7668502de67afbaae07130088da5c6f61e45/example/example.cpp#L181
Replacing `ads-server` with `192.168.0.231` should work for the setup you described.

ZhousuLee, 2018-12-03:
You solved my problem perfectly .I forget this step:
> Replacing `ads-server` with `192.168.0.231` should work for the setup you described.
Thank you very much.

closed 2018-12-06 by ZhousuLee