Beckhoff/ADS #83: ADS for .NET and Mono

robertsabo0, 2019-03-27:
Dear,

I'm working on implementation of reading PLC with TwinCAT 2 from .NET application.
On windows, I succeed, and all is good. 
But on Linux (using mono) I cannot connect to PLC.

I installed c++ lib as docs says  (https://github.com/Beckhoff/ADS)

When I run my code I got error that dll is not found:
`Unhandled Exception:
TwinCAT.Ads.AdsInitializeException: ADS could not be initialized because the 'TcAdsDll.dll' is not found! Please check DLL search paths!
(tcadsdll.dll) ---> System.DllNotFoundException: tcadsdll.dll
  at (wrapper managed-to-native) TwinCAT.Ads.Internal.TcAdsDllWrapper+UnsafeNativeMethods.AdsGetDllVersion()`

I guess the reason is that I installed c++ lib for ADS, not .NET.

Am I wrong?
Is it possible to read TwinCAT using .NET from Linux?

pbruenn, 2019-03-27:
Hi Robert,
TwinCAT ADS .NET from:
https://infosys.beckhoff.com/english.php?content=../content/1033/tcadsnet/html/twincat.ads.intro.html&id=
requires
TwinCAT ADS-DLL from your TwinCAT Installation :-(
https://infosys.beckhoff.com/content/1033/tcadsdll2/html/tcadsdll_intro.htm?id=3725064819111272020

The opensource library (AdsLib) you find here on GitHub is intended as a replacement for TwinCAT ADS-DLL, but the feature set is not 100% equal.
The opensource library is limited by:
- no real central AMS router instance is implemented
- for functions with multiple versions only the latest version is implemented (f.e. `AdsSyncReadReqEx2`)
- [AdsNotificationHeader](https://github.com/Beckhoff/ADS/blob/eda55603f88bcbebd397060a02b12377debd2137/AdsLib/AdsDef.h#L374) was reordered.

I don't know the details of TwinCAT ADS .NET, but I am pretty sure the missing router is already a show stopper to make it work with opensource AdsLib.

I know of a Python wrapper which uses AdsLib:
https://github.com/stlehmann/pyads

Maybe you find a .NET fork on github, too:
https://github.com/Beckhoff/ADS/network/members

Regards,
Patrick

arj3090, 2019-07-05:
There is an ADS driver on Sourceforge that uses the TCP/IP stack instead of the AMS router, so it will possibly work on Linux under Mono:

https://sourceforge.net/projects/twincatads/

The catch is the TwinCAT 2 AMS router has a flaw in which it will catch traffic and no matter which TCP/IP port it is received on, it will respond to the standard TwinCAT port, therefore the driver will not work if TwinCAT 2 is installed on the same PC (Of course Linux is not an issue). So far my tests seem to show this flaw does not exist in the TwinCAT 3 router.

closed 2019-09-05 by pbruenn