Beckhoff/ADS #87: AMS route setting on the server

pogojotz, 2019-04-25:
We want to implement a flexible client infrastructure, where new clients can be added without any manual changes to the TwinCat server. What stands in our way right now is the need to configure a AMS route to the client in the TwinCat server.

I know it should be possible to set these routes from a remote location. TwinCat can do that to other TwinCat devices provided it has their credentials.

My question now is, can I realize this also from our Client using AdsLib? If not, could you suggest a way to realize this by other means?

pbruenn, 2019-04-25:
That's a tricky question. Yes, I hate adding routes, too! We both know that it somehow seems to work. But I don't know were it is documented, and I doubt it is.
Try ask Beckhoff support for it, or reverse engineer it and post it to the public. Unfortunately I can't help you with this.

pbruenn, 2019-04-25:
Well, I think this counts as official Beckhoff documentation:
https://download.beckhoff.com/download/Document/product-security/Advisories/advisory-2017-002.pdf

pogojotz, 2019-05-09:
I gave up on this. We go another solution with the default TcAdsDll and a proxy tool in the middle for now. I would have loved to use the OS AdsLib - alone for its clean and somewhat better documented include files, compared the the TcAdsAPI.h.
I hope Beckhoff will spawn more OS projects like this in the future to improve interoperability and code quality.

closed 2019-05-09 by pogojotz