Beckhoff/ADS #89: Port 0x2710 is out of range

user500, 2019-05-03:
If I run the following code:

```
#include "AdsLib.h"
#include "AdsDef.h"

#include <unistd.h>

int main() {
	long error_code;

	AmsNetId ams_address_net_id = {192, 168, 1, 127, 1, 1};

	error_code = AdsAddRoute(ams_address_net_id, "192.168.1.127");
	if (error_code) {
		printf("Error: AdsAddRoute: %li\n", error_code);
		throw std::exception();
	}

	while (1) {
		sleep(1);
	}

	return 0;
}

```
I get:

2019-05-02T18:50:44-0700 Info: Connected to 192.168.1.127
2019-05-02T18:50:45-0700 Warning: Port 0x2710 is out of range
2019-05-02T18:50:45-0700 Warning: No response pending
2019-05-02T18:50:49-0700 Warning: Port 0x2710 is out of range
2019-05-02T18:50:49-0700 Warning: No response pending

The last two warnings continuously repeat.

pbruenn, 2019-05-03:
My bet is there is something on 192.168.1.127 sending out packets on the AMS TCP connection. In best case a TwinCAT trying to reach AMS port 0x2710.

user500, 2019-05-03:
I guess maybe I am a little confused. Shouldn't the TwinCAT PLC that I am reading the ADS symbols of be running on the computer at this IP address? Is this the TwinCAT you are referring to?

pbruenn, 2019-05-04:
Well, Your sample code is empty. All it does is opening a TCP connection to the host running your TwinCAT PLC.
Did you tried the example?

user500, 2019-05-04:
This was a minimal working example that gives the same warnings as code I wrote to read the values from the PLC.

pbruenn, 2019-05-04:
Now you know the PLC or some other ADS device is sending packets over your TCP connection. If you don’t know what service this might be, I would take a wireshark capture and look into these packets.

user500, 2019-05-05:
I'll take a look. Thank you.

user500, 2019-05-07:
Found it. It was the TwinCAT3 Scope. Thank you for your help.

________________________________
From: Thomas, Patrick
Sent: Saturday, May 4, 2019 6:13 PM
To: Beckhoff/ADS; Beckhoff/ADS
Cc: Author
Subject: Re: [Beckhoff/ADS] Port 0x2710 is out of range (#89)

I'll take a look. Thank you.

closed 2019-09-05 by pbruenn

simonschmeisser, 2023-09-25:
This message is somewhat confusing, could it be muted? Do you need a wireshark capture? The message stops arriving every 10s once closing the Visual Studio.

pbruenn, 2023-09-26:
Well, you can hide warnings by setting `Logger::logLevel=4` https://github.com/Beckhoff/ADS/blob/master/AdsLib/Log.h#L26, but you already know why it is happening. Simply because something else on the same machine is sending requests from an "Amsport 0x2710". This thing is most certainly TwinCAT running on the same machine as your ads application. The thing is if you want them to coexists, then stop using the standalone mode and switch to twincat mode e.g. use the existing TwinCAT router on that system. This is done by setting tcadsdll_include and tcadsdll_lib https://github.com/Beckhoff/ADS/blob/master/meson_options.txt

simonschmeisser, 2023-09-26:
Actually we don't have anything besides this library running on our Linux computer. TwinCat IDE is running on the Windows computer. This is not a high priority issue for us, just would be nice to have in order not to confuse users.

I would expect that some warnings would be important enough so I don't want to miss them and disabling them is a bad idea?

pbruenn, 2023-09-26:
hmm, this is something I don't understand. You have a Windows TwinCAT Client, a Linux Client with your AdsLib application and a TwinCAT PLC?  

And then you see these warnings on the linux client? Actually in that case your wireshark capture or a diagram of your setup (with ip addresses) might be helpful. Send me an email if you want. This sounds like a pretty specific issue.

Mebus, 2025-01-09:
Hey,

I am getting the same message and I have also got only one client running on the Linux machine.

Mebus

pbruenn, 2025-01-10:
> Hey,
> 
> I am getting the same message and I have also got only one client running on the Linux machine.
> 
> Mebus

Take a look into wireshark. 0x2710 is 10000 which is the AmsPort of the systemservice. Someone is trying to "respond" to a systemservice on your Linux machine. Maybe some TwinCAT XAE which at some point was connected to a TwinCAT with the same IP as the Linux client, now.