Beckhoff/ADS #93: twincat server on localhost

ao-droid, 2019-08-02:
I would like to run ads clinet on the same machine as the server.  Does this lib support it?

pbruenn, 2019-08-03:
No, use one of the libraries provided with TwinCAT:
https://infosys.beckhoff.com/english.php?content=../content/1033/tcadscommon/html/tcadscommon_intro.htm&id=

ao-droid, 2019-08-03:
I preasume I have to change the example... Since with Beckhoff libs I only need the port number if you connect locally?...in your version I need to specify a static route in TC

pbruenn, 2019-08-03:
Yes, with the TwinCAT dll you don’t need to setup a route, especially on localhost.

ao-droid, 2019-08-03:
OK thanks for replay.

pogojotz, 2019-09-04:
I had it running once on the localhost, but the route "back" has to be configured manually in TwinCat and it sometimes failed to work (don't remember the details).

At some point I gave up due to the lack of reliability and used the blob from Beckhoff.

ao-droid, 2019-09-04:
Thanks for your info...I also used the Beckhoff libs... However recently somebody told me that the git libs also work locally, but I forgot to post it. It turns out you need to guess the client AmsNetId. Only last value needs to be changed... but it only works with specific values. I was told it can also be done without the static route in twincat3 which did't work in my case.  Don't know if it's reliable because I didn't have time for a through test...maybe some other time.

closed 2021-03-16 by pbruenn

ccvca, 2023-10-12:
I've had success using the external IP and using `AdsSetLocalAddress` to set a different ID and add this route to ADS E.g.:
```
AdsSetLocalAddress({192,168,1,23,1,2});
AdsDevice route {"myDNSForADS", {192,168,1,23,1,1}, AMSPORT_R0_PLC_TC3};
```