Beckhoff/ADS #94: Build error when compiling in Visual Studio 2019

nouvak, 2019-08-13:
I noticed I get the following error when building ADS in Visual Studio 2019 (Community Edition):

```
FAILED: CMakeFiles/AdsClient.dir/Sockets.cpp.obj
C:\PROGRA~2\MICROS~4\2019\COMMUN~1\VC\Tools\MSVC\1422~1.279\bin\Hostx64\x64\cl.exe  /nologo /TP -DADSCLIENT_BUILD_LIBRARY -DAdsClient_EXPORTS -I. -I..\..\..\ -I\common\install\include /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MD /O2 /Ob2 /DNDEBUG /showIncludes /FoCMakeFiles\AdsClient.dir\Sockets.cpp.obj /FdCMakeFiles\AdsClient.dir\ /FS -c ..\..\..\Sockets.cpp
C:\Users\user\git\common\ADS\AdsLib\Sockets.h(45): error C2039: 'runtime_error': is not a member of 'std'
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.22.27905\include\string(19): note: see declaration of 'std'
C:\Users\user\git\common\ADS\AdsLib\Sockets.h(45): error C2504: 'runtime_error': base class undefined
C:\Users\user\git\common\ADS\AdsLib\Sockets.h(46): error C2039: 'runtime_error': is not a member of 'std'
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.22.27905\include\string(19): note: see declaration of 'std'
C:\Users\user\git\common\ADS\AdsLib\Sockets.h(46): error C2614: 'Socket::TimeoutEx': illegal member initialization: 'runtime_error' is not a base or member
..\..\..\Sockets.cpp(112): warning C4267: 'argument': conversion from 'size_t' to 'int', possible loss of data
..\..\..\Sockets.cpp(120): warning C4996: 'strerror': This function or variable may be unsafe. Consider using strerror_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt\string.h(182): note: see declaration of 'strerror'
..\..\..\Sockets.cpp(155): warning C4996: 'strerror': This function or variable may be unsafe. Consider using strerror_s instead. To disable deprecation, use _CRT_SECURE_NO_WARNINGS. See online help for details.
C:\Program Files (x86)\Windows Kits\10\include\10.0.18362.0\ucrt\string.h(182): note: see declaration of 'strerror'
..\..\..\Sockets.cpp(171): warning C4267: 'argument': conversion from 'size_t' to 'int', possible loss of data
[12/20] Building CXX object CMakeFiles\AdsClientStatic.dir\AmsRouter.cpp.obj
ninja: build stopped: subcommand failed.
```

The fix is to include the "<stdexcept>" header in "ADS/AdsLib/Sockets.h":

#include "Frame.h"
#include "wrap_socket.h"
#include <exception>
+ #include <stdexcept>
#include <string>

pbruenn, 2019-08-13:
Could you please clean up your comment? I can't read what header you mean

pogojotz, 2019-09-04:
He means \"\<stdexcept\>\" (apparently forgot to escape the \< and \>).
Source:
https://api.github.com/repos/Beckhoff/ADS/issues/94

closed 2019-09-05 by pbruenn

nouvak, 2019-09-05:
Oh, sorry! I totally overlooked the formatting issue. :(
As @pogojotz correctly stated, the "<stdexcept>" include was missing..

commit 6b3a0300 (2019-09-05): Socket: include stdexcept instead of exception

fixes #94