Beckhoff/ADS #95: Twincat 3 ADS switching from Config to Run Mode Error

jeffsyang, 2019-08-19:
OS: Windows 10 Pro 64-bit (Build 1809)
Version: TwinCAT 3.1.4020.32 which includes TwinCAT System Service v3.1.0.2211

When trying to switch from Config mode to Run mode via the taskbar icon, TcSysSrv throws an Event 14 error which Event Viewer details as 

> Error starting TCRTSOBJECTS server. Win32 error: 3 (The system cannot find the path specified). 

I have tried running `C:\TwinCAT\3.1\System\win8settick.bat` as admin without any success. It also seems like other servers such as TCIO, TCRTime, TcEventLogger, ran successfully. Although this might not be the best place to ask, I was wondering if there's something I'm missing?

Edit: Just noticed that my syssrv exe was named `C:\TwinCAT\3.1\System\TCATSysSrv.exe`; is that correct? Is it supposed to be `TcSysSrv.exe`?

jeffsyang, 2019-08-19:
[image]
Maybe I'm missing something here? If so, how would I install it?

pbruenn, 2019-08-20:
Sorry, this is absolutely the wrong place to ask. I have no idea what you are talking about. please contact your TwinCAT support, they should be able to help you pretty fast.

closed 2019-08-20 by pbruenn