Beckhoff/ADS #97: Port 0x353 is out of range (only on linux)

michieldwitte, 2019-08-21:
I have a working example on windows to read out a variable by name.
I can not seem to get this to work on linux.
The contents of the requests looks the same in wireshark.
The PLC echo's the Linux AMS Request, the windows request receives a valid answer.

I've tried using the example.cpp and the pyads wrapper.
This leads to the same output:

```
2019-08-21T14:47:32+0200 Info: Connected to 10.0.0.48
2019-08-21T14:47:32+0200 Warning: Port 0x353 is out of range
2019-08-21T14:47:32+0200 Warning: No response pending
```

[wireshark captures.zip](https://github.com/Beckhoff/ADS/files/3525409/wireshark.captures.zip)

Am I missing something here?

pbruenn, 2019-08-21:
Your linux.pcapng looks weird. AMS NetId and Port should change between sender and target but in both of your packets they keep the same, even if the IP and MAC address changes.
Could you elaborate a little more about your setup?

michieldwitte, 2019-08-21:
I rechecked the routing and manually re-entered it and it solved the problem, thanks and sorry for wasting your time.

closed 2019-08-21 by michieldwitte