Beckhoff/ADS #98: Cannot add a route for my ADS client on Ubuntu

Drikuy, 2019-09-18:
Hi,

I'm trying to do an PC control application on Ubuntu and I want to use ADS connection to communicate with a PC running TC3. The thing is that I've followed the configuration steps but when I come to adding a route TC3 system shows me an 1800 error and do not add the route. So when I try to make the connection, as the TC3 does not save the static route to Ubuntu running on a Virtual Machine, the connection closed by remote. 

I've configured the AmsNetId on Ubuntu app as the AmsNetId of the TC3 system and remoteIpV4 too. I'm also uncommenting AdsSetLocalAddress() and giving the a.b.c.d.1.1 value where a.b.c.d is my Ubuntu machine IP. Am I making a mistake on the Ubuntu example.cpp configuration?

I believe that the error is because TC3 does not allow me to add the static route. Here's a picture so you can see the message TC3 shows me:

[image]

What I'm doing wrong in the TC3 route configuration?
Thanks in advance for your help.

pbruenn, 2019-09-18:
Select „None / Server“ in the „Remote Route“ Radio box

Drikuy, 2019-09-18:
I've tryed what you told me to do and there's no success. Now I've get an TimeOut error 1861 as you can see in the following picture.

[image]

I've already check out Windows and Ubuntu firewalls status and the ports are opened. But the TC3 system still does not found Ubuntu System (I can ping from Windows to Ubuntu and vice versa). Should I run `./example.bin` on Ubuntu while TC3 system is searching through ADS net?

pbruenn, 2019-09-18:
oh and select "IP Address" instead of "Host Name"

Drikuy, 2019-09-19:
I've select "IP Address" instead of "Host Name" and it still not working.  Do I might have a wrong versión of TC3 installed on my computer? I have installed TC3 XAE build 4024. Do I have to test it with a runtime version instead?

Or maybe I'm making a mistake while trying to configure the route. I am working with TC3 in `config mode`, should I change into `run mode`?

pbruenn, 2019-09-21:
Did you get it to run? The only thing we might miss is the windows firewall.

Drikuy, 2019-09-22:
It still doesn't work. I've also think it could be something to do with windows firewall. But I've tried to full disable it, and it does not connect to ADS server. I've going to post the message I've seen in the console while running `./example.bin` and the configuracion of the `example.cpp` file I've done, maybe this helps you seen what might I be doing wrong:

./example.bin console messages:
```
~/ADS/example$ ./example.bin
2019-09-22T13:51:56+0200 Info: Connected to 192.168.0.20
Adding ADS route was done
Open port is number:30000
2019-09-22T13:51:56+0200 Info: connection closed by remote
Add device notification failed with: 1861
notificationByNameExample():
Create handle for 'MAIN.byByte[4]' failed with: 1861
Add device notification failed with: 1861
readExample():
ADS read failed with: 1861
readByNameExample():
Create handle for 'MAIN.byByte[4]' failed with: 1861
Unable to determine symbol size, reading ADS symbol information failed with: 1861
```

example.cpp file:
```
void runExample(std::ostream& out)
{
    static const AmsNetId remoteNetId { 172, 16, 7, 100, 1, 1 };  // 172,16,7,100, 1, 1 
    static const char remoteIpV4[] = "192.168.0.20";	//172.16.7.100192.

    // uncomment and adjust if automatic AmsNetId deduction is not working as expected
    AdsSetLocalAddress({ 192, 168, 114, 129, 1, 1 });  //  {172, 16, 7, 150, 1, 1}

    // add local route to your EtherCAT Master
    if (AdsAddRoute(remoteNetId, remoteIpV4)) {
        out << "Adding ADS route failed, did you specify valid addresses?\n";
        return;
    }
    else
	out << "Adding ADS route was done\n";

    // open a new ADS port
    const long port = AdsPortOpenEx();
    if (!port) {
        out << "Open ADS port failed\n";
        return;
    }
    else
    	out << "Open port is number:" << std::dec << port << '\n';

    const AmsAddr remote { remoteNetId, AMSPORT_R0_PLC_TC3 }; //AMSPORT_R0_PLC_TC3
    notificationExample(out, port, remote);
    notificationByNameExample(out, port, remote);
    readExample(out, port, remote);
    readByNameExample(out, port, remote);
    readStateExample(out, port, remote);
    out << "Se han leido los valores y se ha debugueado\n";

    const long closeStatus = AdsPortCloseEx(port);
    if (closeStatus) {
        out << "Close ADS port failed with: " << std::dec << closeStatus << '\n';
    }

#ifdef _WIN32
    // WORKAROUND: On Win7 std::thread::join() called in destructors
    //             of static objects might wait forever...
    AdsDelRoute(remoteNetId);
#endif
}
```

Should I've Install any executable on Ubuntu side in order to simulate an ADS net in order to get it recognize by Windows TC3? Thanks for your help.

pbruenn, 2019-09-22:
Oh, I think I mixed up something. So you get the error when running the example not when adding the route?
In case you successfully added a route on the TwinCAT side. And all the addresses match together, than you indeed have to put TwinCAT to RUN mode in order to communicate with the PLC.

Drikuy, 2019-09-22:
I get error when adding the route, but I was wondering if this could be because of a missconfiguration of Ubuntu client. The point as I told before is that I cannot add any non-Beckhoff device to the ADS router in my TC3 running on windows. Sorry about my last message if there's something that make you missunderstand my problem.

Drikuy, 2019-09-23:
I think I've achieved something. I was making a mistake of not opening Ubuntu Firewall while trying to add ADS route into my Windows PC running TC3. And when I add a the new route, route wizzard make something wierd: it does not show an error message but it does not show the route. It is empty:

[image]

So I have the firewall disabled in both sides and the comunication still shows me (on Ubuntu side): `connection closed by remote` and does not allow me to add a route into the PLC side.

pbruenn, 2019-09-24:
I returned to the office and can confirm your issue with XAE 4024.0. My colleagues confirmed they are working on a fix for upcoming 4024.1
So either use an older version of XAE or you could try to edit C:\TwinCAT\3.1\Target\StaticRoutes.xml by hand:
```
<TcConfig xmlns:xsi="http://www.w3.org/2001/XMLSchema-instance">
	<RemoteConnections>
		<Route>
			<Name>ubuntu</Name>
			<Address>192.168.114.129</Address>
			<NetId>192.168.114.129.1.1</NetId>
			<Type>TCP_IP</Type>
			<Flags>32</Flags>
		</Route>
	</RemoteConnections>
</TcConfig>
```

Drikuy, 2019-09-24:
Thank you very much! It works.

closed 2019-09-24 by Drikuy