Beckhoff/ADS #99: Read type definitions

jsandv, 2019-09-22:
Im trying to read the type definitions from a CX5020. Im stuck, and went to the C# dll for help 
by using this command:
var symbols = SymbolLoaderFactory.Create(client,SymbolLoaderSettings.DefaultDynamic).Symbols;

Using wireshark I see multiple read commands, one is:
IndexGroup:0x0000f00e, IndexOffset: 0x80000000, Length:65536

When I convert the bytes to ascii I see that types and definitions of interest is in the response but I dont know the format, so there is no way I can deserialize it without the spec.

Does anyone know where to find more documentation on the symbols / datatype queries?

dxpke, 2020-03-27:
You may want to install the TwinCAT ADS SDK and look at the `ADSIGRP_SYM_DT_UPLOAD` define and the `AdsDatatypeEntry` structure in `TcAdsDef.h` (it's not open source so I can't copy it here)

pbruenn, 2023-01-13:
This should now be possible see  commit  4cc5ded755f119d431f232502ffbe0688865f5ba and its parents

closed 2023-01-13 by pbruenn