Beckhoff/ADS #101: Error: read frame failed with error: Unknown error

mriepl, 2019-10-07:
When you disconnect the network connection to the remote system during an Ads-Functioncall you got an Ads Error:1861 Timeout. Thats the right behaviour.

When you after the Error 1861 reconnect the network and perform a new Ads-Functioncall the Methode "AmsConnection::Receive" stay stucked because bytesToRead is != 0.

In the socket.read() function bytesRead = -1 and lastError = 10054

I think this is a bug, but im not shure whats the most beautiful way to fix this.
You can add (lastError == CONNECTION_RESET)
and in the wrap_socket.h
#define CONNECTION_RESET WSAECONNRESET.

pbruenn, 2019-10-08:
Sounds reasonable and was already mentioned in #78 . Unfortunately I am pretty short on time. That TryRecv() issue will celebrate one years anniversary, soon ...
The root source for all your trouble seems the bad TCP connection handling. My idea would be to move socket.Connect() from AmsConnection constructor into TryRecv() and call it in a loop there.
That would mimic the behavior of TwinCAT SystemService, which continuously tries to reconnect every five or so seconds.

mriepl, 2019-10-08:
Hy Patrick,

i have done a lot of changes to get the reconnection, i am sure this will not be the best implementation but maybe you can reuse some snippet.

[AdsLib.zip](https://github.com/Beckhoff/ADS/files/3704835/AdsLib.zip)

closed 2020-02-03 by mriepl