Beckhoff/ADS #104: ADS error getting handle

marcoojer, 2019-11-13:
Hi,  I'm using the AdsLib in windows 10 with Twincat v3.1.4022.30. I'm trying to connect to a the local (virtual) PLC. I use this code
```
static const AmsNetId remoteNetId{ 192, 168, 100, 16, 1, 1 };
	static const char remoteIpV4[] = "192.168.100.16";
	if (AdsAddRoute(remoteNetId, remoteIpV4)) {
		out << "Adding ADS route failed, did you specify valid addresses?\n";
		return;
	}

   long port = AdsPortOpenEx();
  
    if (!port) {
        out << "Open ADS port failed\n";
        return;
    }
	AmsAddr Addr{ remoteNetId, AMSPORT_R0_PLC_TC3 };

```
the console outputs Connected to 192.168.100.16
and when I try to read the handle it throws the ADS 1861 error.

I don't know if because is windows 10 and the same ip of the machine, it is necessary to open a AMS route. Anyway I did it, but it is still not working. I create the route in the following way.
[image]

Anyone knows  why is it failing?

pbruenn, 2019-11-14:
Please see issue #93. It sounds as it's the same problem.

marcoojer, 2019-11-14:
ok, thank you!!

marcoojer, 2019-11-14:
Hi again, now I'm trying to do the same with two different laptops, both with windows 10. From the target one which runs the PLC project, I can see the remote route in edit route dialog. but when I run the program in the remote i obtain the following:
```
Info:Connected to 192.168.6.30
port:30000
Create handle for GVL.iCount failed with 1861
ADS read failed with 1861
Warning:Port 0x2710 is out of range
```
What could it be the problem?

In the remote computer, I only install the twincat ADS package,

pbruenn, 2019-11-15:
Do you have the route on both of your laptops?
Did you allow ADS communication in your firewall?
If you use Windows, it might be easier to just use the TwinCAT TcAdsDll.dll:
https://infosys.beckhoff.de/english.php?content=../content/1033/tcconnectivity/html/tcconnectivity_intro.htm&id=

kapkunal, 2019-11-23:
I get the same error. Create handle for failed with 1861.
My system is Ubuntu 18 gcc 7.4
It seems to work fine Ubuntu 16 gcc 6.5

pbruenn, 2023-01-12:
Closed as it seems outdated

closed 2023-01-12 by pbruenn