Beckhoff/ADS #108: ADS-State Notification invalid value

mriepl, 2020-01-29:
Hi,

if i add a notification to the ads-state:
[image]
i get the notification and an handle (see line 1):
[image]

I also received the initial notification after the add call.
In the notification header i received  this:
[image]

The given value seems to be wrong.
With the TcAdsDll everything work fine.

pbruenn, 2020-01-31:
Do you use the TcAdsDll header? It might be related to issue #12 .
See my old [comment](https://github.com/Beckhoff/ADS/issues/12#issuecomment-148304863) for details.
If you want to use the same header, you can try a patch like this:
https://github.com/dabrowne/ADS/commit/e181ca260d5cdcdd7afc4bf07c80393f4ce96ddc

mriepl, 2020-02-03:
In my wrapper i have added to the struct AdsNotificationHeader the entry "data[ANYSIZE_ARRAY]" like TcAdsDll.

The problem was that the  notification datas was copied after the entry "data[ANYSIZE_ARRAY]" because in the marked line you want to jump to the data section of the callback with pointer arithmetic.

[image]

closed 2020-02-03 by mriepl