Beckhoff/ADS #110: Improve documentation

jenschr, 2020-02-18:
I'm learning how to use ADS and I have a bit of C++ code that works well, but when I transfer it to C# I'll get the error "Ads-Error 0x701 : Service is not supported by server". Beckhoff's website lists these errors, but neither of them contain links that would explain WHY one would receive such an error. As mentioned - the very same source fails in C# and works flawlessly in C++. Without better documentation, it is impossible to actually resolve this error.

Beckhoff support just threw their hands in the air and said "we have a sample code that works" - without looking at the actual problem of the error message, so without better documentation, it's really hard to be a Beckhoff customer?

soberschmidt, 2020-02-18:
Hi Jens,
lets start with some background information:
For C# Beckhoff provides the TwinCAT.Ads.dll which wraps the TcAdsDll.dll, which is the used for C++ Application on TwinCAT Systems. The ADS Client API of this GitHub is intended for NON-TwinCAT systems.
Nevertheless ADS functionality which is used in C++, should also work with C#, because it ends in the same same ADS request. Are you able to share your code and some target information of the ADS server (TwinCAT Version and which ADS device).

Best Regards,
Sven

jenschr, 2020-02-18:
Hi Sven,
At the point I got that error, I had a very simple C# program that was based off one of the C++ examples. My mistake was that when I moved it to C#, I noticed that there was an event called "AdsStateChanged" so I added it. Apparently, adding a listener for this event will make a TcAdsClient throw that error once you try to connect the client. Once I removed the listener, it worked as intended. I can't see how that error message is related to that event? Could it be that it trickles down and is caught in the wrong try/catch block?

I can dig back to the code I used at that point if you want to have a look.

I just found a somewhat similar issue now with "ConnectionStateChanged" (also on the TcAdsClient). Adding a listener to this object will not Dispatch an event upon change, but the "isConnected" property will update as it should.

I can work around these things, but I find it odd that the events are exposed if they don't work as intended.

pbruenn, 2020-02-19:
Hi Jens,
yes this is the wrong place for C# issues, but since you finally found someone to listen to you, please use the chance and describe your problem. So Sven and other experts can help you.

jenschr, 2020-02-24:
Thanks Patrick,
I made a brief WPF example that shows the problem. Adding an event listener for AdsStateChanged in this example (line 50) http://flashgamer.com/a/WpfApp1.zip will give you a crash with a non-relevant error. Removing it will solve the crash, but you won't get notifications for state change.

If the documentation contained any info, I could use it to solve the issue. Maybe I cannot do this until I'v done X? I have no way of knowing if I've discovered a bug or if this is expected behavior. Really good documentation offers examples of how to use a feature, but the Beckhoff documentation is generally just generated from the source code with no additional info? Here's the entry for the AdsStateChanged Event https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_adsnetref/7312916619.html&id=

It basically tells me nothing that I cannot get from the intellisense feature in VS?

soberschmidt, 2020-03-09:
Hi Jens,
your example contains no relevant ADS logic. I can start it without any issue. Maybe you uploaded the wrong one. I have tested the Eventhandler again with the latest TwinCAT 3 PLC to detect state changes, which works without issues.
If you want to detect state changes of the TwinCAT system service, to check if TwinCAT is in config or run-mode, you can use this sample: https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_adssamples_net/185256971.html&id=1261277081080414733
The TwinCAT.ADS.dll is not part of this repository and will be supported by Beckhoff. Please contact the support directly by sending a mail to support@beckhoff.com.

Thanks Sven

closed 2020-03-09 by soberschmidt