Beckhoff/ADS #112: Improve Documentation on Error Codes

Reda1000, 2020-04-21:
Please add more details on troubleshooting problems based on ADS Error Codes on official ADS Error Code lookup table such as: https://infosys.beckhoff.de/index.php?content=../content/1031/tcdiagnostics/html/ads_returncodes.htm&id=5768364593865330432

I spend some hours fiddling around with a TcAdsWebservice (Javascript) + IIS-AdsWebserviceDLL Setup and just by luck found this issue adressing my Problem: https://github.com/Beckhoff/ADS/issues/39

As already mentioned in ticket above it is not trivial to guess that 1793 can be caused by misconfigured ports when ADS Error Code table just mentions "Service is not supported by server" -  to be fair I am a beginner but I most likely face such issues again and again by the fact that there are TCP-Port, AMS-Port, PLC-Port etc. that can be messed up.

@pbruenn you doing a good Job :)

soberschmidt, 2020-09-21:
The AMS and ADS error are generic, because they are used by all TwinCAT servers/services. In my opinion "Service is not supported by server" is clearly. More samples related to the different ADS access could fit better to address that topic. In general I would recommend to work with the symbols instead of indexgroups and offsets like it is shown in the existing samples https://infosys.beckhoff.de/content/1031/tcsample_tcadswebservicejs/html/sample01.html?id=3681266362023774572

Reda1000, 2020-09-21:
Hi @soberschmidt , thanks for your comment. So I understand that there is no intention on adding more information i.e. to column "Possible Causes" and "Solution" in der ADS Error Code Tables.

I will close this as I got around my initial issue. Please be aware though that I still feel there should be more information given stating what possible reason caused a Error - even or especially when the ErrorCodes are generic thus even more complicated to understand in specific situations.

closed 2020-09-21 by Reda1000