Beckhoff/ADS #122: Add an example to read WSTRING with OOI

Frants1987, 2020-12-02:
Hello,
I started to build an application using the OOI version of the lib. However, I'm missing an example on how to read strings / wstrings from TwinCAT.
I tried using:
-  `AdsVariable<std::string>` which gave me nothing useful since the string has no defined size
-  `AdsVariable<std::array<char, 80>>` which wasn't helpful either
Could anyone point me in the right direction on how to use this class?
Thanks

Frants1987, 2020-12-03:
OK, I got it to work with `AdsVariable<std::array<unsigned short, 80>>` but please point me to a better solution if there is one.
Btw. thanks for making the code available to the public.

pbruenn, 2020-12-03:
I don't know any better solution. This might be helpful:
```
using WSTRING = std::array<uint16_t, 80>;
AdsVariable<WSTRING> var;

```

closed 2020-12-03 by Frants1987