Beckhoff/ADS #123: ADS timeout  1861 connectiong Raspberry Pi to CX9020

ideasoft-bz, 2020-12-08:
Hello,
I am trying to communicate from Raspberry Pi to CX9020 running TC2. I get the timeout 1861 error:

./testconn.bin
2020-12-06T18:23:48+0100 Info: Connected to 192.168.1.30
2020-12-06T18:23:53+0100 Error: select() timeout
2020-12-06T18:23:53+0100 Warning: InvokeId of response: 0 timed out
Error: 1861
AdsException message: Ads operation failed with error code 1861.
2020-12-06T18:23:53+0100 Info: connection closed by remote

I carefully read issue #14 adn my code is according to suggestion there:

#include "AdsLib.h"
#include "AdsVariable.h"
#include <iostream>
int main()
{
static const AmsNetId remoteNetId { 5, 84, 174, 32, 1, 1 };
static const char remoteIpV4[] = "192.168.1.30";
AdsSetLocalAddress({192, 168, 1, 3, 1, 1});
if (AdsAddRoute(remoteNetId, remoteIpV4)) {
std::cout << "Adding ADS route failed\n";
}
try {
AdsDevice route {remoteIpV4, remoteNetId, 801};
//const auto state = route.GetState();
const auto info = route.GetDeviceInfo();
}
catch (const AdsException& ex) {
std::cout << "Error: " << ex.errorCode << "\n";
std::cout << "AdsException message: " << ex.what() << "\n";
}
catch (const std::runtime_error& ex) {
std::cout << ex.what() << '\n';
}
}

From the same environment, python program using pyads works:

Python 3.7.3 (default, Jul 25 2020, 13:03:44) 
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pyads
>>> plc = pyads.Connection('5.84.173.32.1.1', 801, "192.168.1.30")
>>> plc.open()
2020-12-08T11:57:31+0100 Info: Connected to 192.168.1.30
>>> print (plc.read_device_info())
('TCatPlcCtrl', <pyads.structs.AdsVersion object at 0x766a1530>)
>>> plc.close()
2020-12-08T11:58:03+0100 Info: connection closed by remote

Any suggestion please?

pbruenn, 2020-12-10:
`2020-12-06T18:23:53+0100 Warning: InvokeId of response: 0 timed out Error: 1861` is an indicator that you receive data from another session. Do you run some other ADS client in parallel?
Normally I would suspect TwinCAT, but I don't believe you run TwinCAT XAE on your Raspberry Pi

ideasoft-bz, 2020-12-10:
Yes, I have route to rasPi. 
From raspi I am able to ping CX9020 IP address, but I am not able to ping its hostname. Is it possible that this is problem?

[image]

pbruenn, 2020-12-10:
static const AmsNetId remoteNetId { 5, 84, 174, 32, 1, 1 };
doesn't match your pyAds config
plc = pyads.Connection('5.84.173.32.1.1', 801, "192.168.1.30")

ideasoft-bz, 2020-12-10:
Yes, this is something I checked so many times and I newer really saw any difference.

Thank you very much

closed 2020-12-17 by ideasoft-bz