Beckhoff/ADS #124: Need help with AdsVariable

ideasoft-bz, 2020-12-21:
Hello,

My C++ knowledge is not good enough to decode  AdsVariable. In the example 

 static void readByNameExample(std::ostream& out, const AdsDevice& route)
 {
    AdsVariable<uint8_t> readVar {route, ".StevPalet"};

    out << __FUNCTION__ << "():\n";
    for (size_t i = 0; i < 8; ++i) {
        out << "ADS read " << std::hex << (uint32_t)readVar << '\n';
    }
 }

This example works, but how do I decode return data  if my request is for variable type of bool, int, real?

Thank you very much for help

Regards
Bojan

Frants1987, 2020-12-23:
Hello Bojan,
it should look something like this:
* `AdsVariable<bool> readVarBool {route, ".myBool"};` for type BOOL
* `AdsVariable<int16_t> readVarInt {route, ".myInt"};` for type INT
* `AdsVariable<int32_t> readVarDInt {route, ".myDInt"};` for type DINT etc.
* `AdsVariable<float> readVarReal {route, ".myReal"};` for type REAL

Just make sure the size of the variables is the same in TwinCAT and on your platform. You can find more information on type sizes [here](https://en.wikipedia.org/wiki/IEC_61131-3).
You can assign the value of the variable using the assignement operator:
`bool value = readVarBool;`
Hope I could help you.

ideasoft-bz, 2020-12-23:
Super, thank you very much.

Regards Boajn

closed 2020-12-23 by ideasoft-bz