Beckhoff/ADS #125: Definition of ads_i32

Frants1987, 2020-12-23:
Hello,
I'm trying to build the AdsLib against the TwinCAT library but I'm stuck with the definition of the `ads_i32` type.
It always gives me the following errors:
"Error	C2666	'AdsSyncDelDeviceNotificationReqEx': 2 overloads have similar conversions"
"Error	C2666	'AdsSyncWriteControlReqEx': 2 overloads have similar conversions"
"Error	C2666	'AdsSyncWriteReqEx': 2 overloads have similar conversions"
if I defined it like this: `typedef long ads_i32;`.
Could you give me a hint on how to circumvent this?
I'm using VisualStudio 2019 with the C++14 compiler.

pbruenn, 2021-01-04:
I don't have a Windows machine at hand but ads_i32 should be either defined as LONG or int on your platform

Frants1987, 2021-01-04:
Yes, I even had to add another enum in "Twincat/AdsDef.h":
```
enum nSystemServiceIndexGroups : uint32_t {
    SYSTEMSERVICE_FOPEN = 120,
    SYSTEMSERVICE_FCLOSE = 121,
    SYSTEMSERVICE_FREAD = 122,
    SYSTEMSERVICE_FWRITE = 123,
    SYSTEMSERVICE_FDELETE = 131,
};

using ads_i32 = long;
```
However since most of the names in AdsLib are equal to the proprietary library I had to introduce namespaces to get is to work. 
I can open a pull request if you are interested in the solution.

pbruenn, 2021-01-04:
Okay, now I think I know what happened. From my example it isn't obvious how to link against TwinCAT. Take a closer look into this commit https://github.com/Beckhoff/ADS/commit/fb39240a0d81dff43aee360c93d9eb71c3cc368f. Especially into the Makefiles.
Since that commit it should be easier to use our object oriented API with the classic TwinCAT router on Windows (or TC/BSD) platforms. What changed is we now have to different libs: AdsLib-win32.a and the new TcAdsLib-win32.a which has the same API like before, but on TwinCAT platforms you can use the TwinCAT router as a backend. However you still have to select your backend at link time.
To use the TwinCAT backend you have to link TcAdsLib-*.a, add the path to your TwinCAT headers to the include path and compile your program with USE_TWINCAT_ROUTER defined.

Frants1987, 2021-01-04:
OK, now I get it. I have no mingw installed on my system and used the visual studio compiler with the .sln file.
Just for clarification, why don't you use namespaces? Just for the backwards compatibility?
Thanks for the help!

pbruenn, 2021-01-04:
Yes, you should be able to apply the changes from the Makefile to your sln.
The next think on my list is an improved example. Unfortunately I didn't found the time yet.
Namespaces pop up quite often. Yes, we started without them because of compatibility. But I see the need for them and will try to add them. Maybe I can convince other to improve the proprietary library , too.

closed 2021-01-04 by Frants1987