Beckhoff/ADS #126: Notification Calling-Convention stdcall vs. __cdecl

mriepl, 2021-01-08:
Hello,

i have seen now it is possible to use the library standalone and with TwinCat Router. 
A view years ago i have wrote an wrapper for exact this case and the notifications maked some troubles.
Because i dont care about the calling convetion of the static callbackfunction.

TwinCat needs stdcall and the AdsLib uses __cdecl calling-convetion.

In the Make file of the example i see you compile with -DUSE_TWINCAT_ROUTER and the NotifyCallback is decleared with __cdecl calling convention. 

I think this would be a problem, but i am not sure :)

Best Regards
Markus

pbruenn, 2021-01-08:
Thank's for the hint please keep this issue open. The new USE_TWINCAT_ROUTER feature definitely needs more testing and I will add more tests and examples. But before I have to do some major changes to our internal infrastructure and that will take some time, which I currently cannot spend.

pbruenn, 2021-03-16:
On win32 __stdcall is the default calling convention, so no need to define it explicitly.
On TC/BSD it's declared as TCADSDLL_API, which itself is defined to nothing, so the standard calling convention is used, just like in our AdsLib.
I think we should be fine. If you have a concrete example, please reopen.

closed 2021-03-16 by pbruenn