Beckhoff/ADS #129: about reading and writing operations - ADS/Linux.

techaitr, 2021-02-10:
Hi,

**_prior knowledge:**
I use windows plc and linux computer. This system was runned using example/example.cpp so, there is not problem about connection._ 

I want to read random float, int, bool variables from plc and I need to change these values. Is there any documentation or simply sample code on this subject?

Thanks.

pbruenn, 2021-02-10:
Well, did you look into example.cpp?
This is how to declare a variable you want to access over ADS using the variables PLC name:
https://github.com/Beckhoff/ADS/blob/ba29306405ff9535ece1410ae4a55e68a8223760/example/example.cpp#L67

So for REAL try
`AdsVariable<float> yourVar {route, "MAIN.yourREAL"};`
or for LREAL try
`AdsVariable<double> yourVar {route, "MAIN.yourLREAL"};`
or for BOOL try
`AdsVariable<uint8_t> yourVar {route, "MAIN.yourBOOL"};`

techaitr, 2021-02-10:
thanks, I looked and understood the example.cpp code but I have another questions. Can I use this library in qt creator?

pbruenn, 2021-03-16:
Why shouldn't you? It's a plain c++11 library with few dependencies, so you should be able to integrate it with qt creator

closed 2021-03-16 by pbruenn