Beckhoff/ADS #131: Are RPC Methods available using the C DLL?

chrisbeardy, 2021-02-13:
Hello, does anyone know if using the C DLL it is possible to do RPC method call of Methods from PLC FB's, like you can in the [.NETimplementation](https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_ads.net/9516517131.html&id=8452097957037553814)?

I have had a look around the C files and can't see anything obvious, I'm not sure if it is possible to get the symbol of the method and just do a read/write to the correct memory areas.?

I ask this as I would like to be able to add this feature into the popular pyads package.

dayaftereh, 2021-02-18:
Hey,

a working example to call a method can be found at [infosys](https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_ads_intro/99079192176427915.html&id=8723761626320715598). 

Example:
```cpp
typedef struct
{
    unsigned long a;
    unsigned long b;
} MethodParameters;

// ...

long nErr, nPort;
AmsAddr Addr;
PAmsAddr pAddr = &Addr;
ULONG lHdlVar, rValue;
char szVar[] = {"MAIN.foo#DoAdd"};
MethodParameters params;

// ...

nPort = AdsPortOpen();
nErr = AdsGetLocalAddress(pAddr);
if (nErr)
{
    cerr << "Error: AdsGetLocalAddress: " << nErr << '\n';
}

pAddr->port = 851;

nErr = AdsSyncReadWriteReq(pAddr, ADSIGRP_SYM_HNDBYNAME, 0x0, sizeof(lHdlVar), &lHdlVar, sizeof(szVar), szVar);
if (nErr)
{
    cerr << "Error: AdsSyncReadWriteReq: " << nErr << '\n';
}

params.a = 5;
params.b = 5;

nErr = AdsSyncReadWriteReq(pAddr, ADSIGRP_SYM_VALBYHND, lHdlVar, sizeof(rValue), &rValue, sizeof(params), &params);
if (nErr)
{
    cerr << "Error: AdsSyncReadWriteReq: " << nErr << '\n';
}

cout << "rValue: " << rValue;

nErr = AdsSyncWriteReq(pAddr, ADSIGRP_SYM_RELEASEHND, 0, sizeof(lHdlVar), &lHdlVar);
if (nErr)
{
    cerr << "Error: AdsSyncWriteReq: " << nErr << '\n';
}

nErr = AdsPortClose();
if (nErr)
{
    cerr << "Error: AdsPortClose: " << nErr << '\n';
}
```

chrisbeardy, 2021-02-18:
this looks great, thanks. Can you repost the link please, its linking to:

[image]

and I cant seem to find that example

[image]

dayaftereh, 2021-02-18:
Yes:

https://infosys.beckhoff.com/english.php?content=../content/1033/tc3_adssamples_net/18014399555831179.html&id=713341834445367571

chrisbeardy, 2021-02-18:
awesome thank you

closed 2021-03-16 by pbruenn