Beckhoff/ADS #135: ninja build issue

mbollini, 2021-03-17:
Hi, I did a clean git.

Meson works fine. This is its output:

```
C:\Users\Maurizio\Documents\GitHub\ADS>meson build
The Meson build system
Version: 0.57.1
Source dir: C:\Users\Maurizio\Documents\GitHub\ADS
Build dir: C:\Users\Maurizio\Documents\GitHub\ADS\build
Build type: native build
Project name: AdsLib
Project version: 0.1
C++ compiler for the host machine: c++ (gcc 9.2.0 "c++ (MinGW.org GCC Build-2) 9.2.0")
C++ linker for the host machine: c++ ld.bfd 2.32
Host machine cpu family: x86
Host machine cpu: x86
Library ws2_32 found: YES
Run-time dependency threads found: YES
Build targets in project: 3

Found ninja-1.10.2 at "C:\Program Files\Meson\ninja.EXE"
```
Ninja build fails because of a lot of errors. 

I report just the very first here:

`C:\Users\Maurizio\Documents\GitHub\ADS>ninja -C build
ninja: Entering directory `build'
[4/17] Compiling C++ object libAdsLib.a.p/AdsLib_Log.cpp.obj
**FAILED**: libAdsLib.a.p/AdsLib_Log.cpp.obj
"c++" "-IlibAdsLib.a.p" "-I." "-I.." "-I..\AdsLib" "-I..\tools" "-fdiagnostics-color=always" "-pipe" "-D_FILE_OFFSET_BITS=64" "-Wall" "-Winvalid-pch" "-Wnon-virtual-dtor" "-Wextra" "-Wpedantic" "-g" -MD -MQ libAdsLib.a.p/AdsLib_Log.cpp.obj -MF "libAdsLib.a.p\AdsLib_Log.cpp.obj.d" -o libAdsLib.a.p/AdsLib_Log.cpp.obj "-c" ../AdsLib/Log.cpp
../AdsLib/Log.cpp: In static member function 'static void Logger::Log(size_t, const string&)':
../AdsLib/Log.cpp:34:9: **error**: 'localtime_s' was not declared in this scope; did you mean 'localtime'?
   34 |         **localtime_s(&temp, TIME_T);** \
      |         **^~~~~~~~~~~**
../AdsLib/Log.cpp:60:9: note: in expansion of macro 'TIME_T_TO_STRING'
   60 |         **TIME_T_TO_STRING(dateTime, &tt);**
      |         **^~~~~~~~~~~~~~~~**`

Any clue?

Regards,

Maurizio.

mbollini, 2021-03-17:
[ninja_build.txt](https://github.com/Beckhoff/ADS/files/6156985/ninja_build.txt)

I started with a clean git, issued 'meson build -Dcpp_std=c++11' but ninja fails to build. See attached log.

pbruenn, 2021-03-17:
Did the old Makefile worked better for you? Sorry I have no other idea than to blame the compiler :-(
https://stackoverflow.com/questions/34847804/stdthread-works-in-cygwin-but-not-in-mingw