Beckhoff/ADS #136: Connection to TC-BDS fails with Error 1827

tillmanott, 2021-03-18:
Hello,

my attempt to use the ADS-lib to connect to Twincat on a CX with TC-BSD fails with error 1827. The readStateExample() and a custom implementation of the library were executed:
[image]

The route between TC-BSD and the Ubuntu machine running the ADS-lib was created. IP address conflicts do not exist. All Ip-addresses were assigned statically in the Class C area within a subnet:
[image]

In TC-BSD, TCP port 48898 was whitelisted in pf.conf and both the firewall and the overall system were restarted.
The following entry was added to pf.conf:
```
#Allow ADS TCP connections 
pass in quick proto tcp to port 48898
```

Disabling the firewall completely did not work either.

I have attached a Wireshark dump, maybe it helps.
[ws_export_adsconnection.zip](https://github.com/Beckhoff/ADS/files/6164125/ws_export_adsconnection.zip)
 
Greetings

pbruenn, 2021-03-18:
What's different to https://github.com/Beckhoff/ADS/issues/128 ?
Your wireshark dump shows "GvlVisu.tstVisuStruct"I expect this to be from your test version. How does the wireshark look like, when you do the readStateExample?

pbruenn, 2021-03-18:
Oh and the addresses in the diagram doesn't match your wireshark dump. In the dump 192.168.215.138 tries to access "GvlVisu.tstVisuStruct", which is your workstation, not the VM

tillmanott, 2021-03-22:
_What's different to #128 ?_
The difference is the operating system of the TwinCat PLC system (in this case TC-BSD). For the windows version of TwinCat everything is working now. The Error in #128 resulted in the previously used link-local-address area. It was resolved by putting everything in a private network (192.168.215.x). Since everything is working fine with the same code and the windows version I assume theres some difference when connectiong to TC-BSD.

_Wireshark dump_
Seems like i attatched the wrong dump. This one should show the readStateExample. The resulting Error 1827 also occur when I use my own code.

[ws_export_error1827.zip](https://github.com/Beckhoff/ADS/files/6182031/ws_export_error1827.zip)

Here's how the example is looking currently:
```
static void runExample(std::ostream& out)
{
    //AmsAddr der PLC. Ermittelt über rechtsklick auf Tray-Icon->Info
    static const AmsNetId remoteNetId { 5, 58, 182, 254, 1, 1 }; //AMS net id of the Beckhoff CX
    //Ip-Adresse der PLC
    static const char remoteIpV4[] = "192.168.215.81"; //ipv4 address of the Beckhoff CX

    //static const AmsNetId remoteNetId { 192,168,232,1,1,1 };
    //static const char remoteIpV4[] = "DERAE-CN217";

    // uncomment and adjust if automatic AmsNetId deduction is not working as expected
    //AdsSetLocalAddress({192, 168, 0, 1, 1, 1});
    AdsSetLocalAddress({192,168,215,138,1,1}); //AMS net id of thhe ubuntu vm (on which the example is running on)
    AdsDevice route {remoteIpV4, remoteNetId, AMSPORT_R0_PLC_TC3};
    //notificationExample(out, route);
    //notificationByNameExample(out, route);
    //readExample(out, route);
    //readByNameExample(out, route);
    //readWriteExample(out, route);
    //readWriteArrayExample(out, route);
    readStateExample(out, route);
}
```

pbruenn, 2021-03-22:
In your diagram 192.168.215.138 is the workstation not the the Ubuntu VM. Are you sure your routing setting on the BSD machine is matching?

tillmanott, 2021-03-22:
Oh, I made a mistake there. The IP addresses are reversed in the diagram. 
Workstation: 192.168.215.137
ubuntu VM: 192.168.215.138

I will adjust the diagram in the first post.

pbruenn, 2021-03-22:
The most interesting part of your edit is that you had the PLC in CONFIG mode. That means you should have received ErrorCode 0x6 and not 0x723.
Sorry, but I have absolutely no idea what's going wrong there.

pbruenn, 2021-07-19:
@tillmanott did you see #148. The problem there was they configured their route as "unidirectional". Could it be your issue, too?

closed 2023-01-11 by pbruenn