Beckhoff/ADS #141: Library running on Beckhoff IPC not working

A-ndre-S, 2021-04-23:
Hi,
I've tryied to compile and use this library in a C++/WinRT project for using WinUI 3 API.
TcAdsDll does not work on this enviroment, giving me 0 when calling AdsPortOpen.
I've followed the example provided with this repo and i get it working when i run the app in my developing PC connected to a Beckhoff IPC running a PLC project. Here is the code:

```
void MainWindow::myButton_Click(IInspectable const&, RoutedEventArgs const&){
	std::string message = "Clicked";
	try {
		static const AmsNetId remoteNetId{ 192, 168, 2, 200, 1, 1 };
		static const char remoteIpV4[] = "192.168.2.200";
		
		// uncomment and adjust if automatic AmsNetId deduction is not working as expected
		AdsSetLocalAddress(AmsNetId{ 172, 22, 50, 158, 1, 1 });
		AdsDevice route{ remoteIpV4, remoteNetId, AMSPORT_R0_PLC_TC3 };
		route.SetTimeout(5000);
		const auto state = route.GetState();
		std::stringstream stream;
		stream << "\tADS state: " << std::dec << (uint16_t)state.ads << " devState: " << std::dec << (uint16_t)state.device;
		message += stream.str();
		AdsVariable<bool> readVar{ route, ".i_main_power_on" };
		message += "\t i_main_power_on: " + std::to_string((bool)(readVar));	
		AdsVariable<int32_t> readenc{ route, ".encoder_AX07" };
		message += "\t encoder_AX07: " + std::to_string((int32_t)(readenc));
	}
	catch (const AdsException& ex) {
		message += "\t";
		message += ex.what();
		std::cout << "Error: " << ex.errorCode << "\n";
		std::cout << "AdsException message: " << ex.what() << "\n";
	}
	catch (const std::runtime_error& ex) {
		std::cout << ex.what() << '\n';
	}
	myButton().Content(box_value(winrt::to_hstring(message)));
}
```

When i try to run the app on the IPC itself i cannot make it working: i can create succesfully the AdsDevice "route" object (following the example) but i get Error 1861 at every call i make.
The changes for the code are on the lines
```
static const AmsNetId remoteNetId{ 192, 168, 2, 200, 1, 1 };
static const char remoteIpV4[] = "192.168.2.200";
		
// uncomment and adjust if automatic AmsNetId deduction is not working as expected
AdsSetLocalAddress(AmsNetId{ 192, 168, 2, 200, 1, 1 });
```
I also tryed to set the remoteIpV4 to localhost.

Every call i make to interact with route (like route.GetState()) raises an exception giving error 1861.

Is it possible to use this library in a PC running a PLC project? Am i setting something wrong?

Thank you,
Andres

pbruenn, 2021-04-23:
On localhost you have to use the TwinCAT router. You, would need to link against TcAdsLib.dll not AdsLib.dll. Call meson with an environment variable pointing to the TwinCAT TcAdsDll.dll similar to this: https://github.com/Beckhoff/ADS/blob/0e0f73f709841c8ee3a0c1ff20ebb46165662d9f/.gitlab-ci.yml#L82

A-ndre-S, 2021-04-25:
Hello,
i've build the library with the option -Dtcadsdll_include=c:\TwinCAT\AdsApi\TcAdsDll\Include, that gives the libTcAdsLib.a. Then i've used this library and TcAdsDll.lib in the same WinUI 3 test project (template with the modifications of the post above) but i have AdsPortOpenEx returning 0 and an exception while calling route.SetTimeout(5000); (code 1864 ADSERR_CLIENT_PORTNOTOPEN).

Have i done something wrong?
Thank you for your help and patience!

Andres

pbruenn, 2021-04-26:
Hmm, if you don't get a valid port from AdsPortOpenEx() maybe you didn't add a TwinCAT route earlier? If you use TcAdsDll directly, does AdsPortOpenEx() work in that case?

A-ndre-S, 2021-04-26:
I've added the route earlier and when using a MFC project TcAdsDll works fine. When i try to use WinUI 3 unluckly i get 0 as return value from AdsPortOpen() and AdsPortOpenEx() using directly TcAdsDll, that's why i tryied to use AdsLib library to connect to a plc running in local. 

(Following "Create a WinUI 3 desktop app for C++/Win32" in https://docs.microsoft.com/en-us/windows/apps/winui/winui3/get-started-winui3-for-desktop)

Thank you!

marcusl, 2022-03-31:
See #161 for some tweaks I've done to use this lib more easily with TwinCAT router on Windows build.