Beckhoff/ADS #145: Improve error handling

simonschmeisser, 2021-05-28:
I'm a total beginner when it comes to ADS and TwinCat in general. I started my exploration based on the example code. Now I get this log output when I try a couple of read operations
```
Starting Beckhoff ADS IO node...
2021-05-28T10:36:12+0200 Info: Connected to 10.2.0.216

## 2x reading
2021-05-28T10:36:16+0200 Warning: Frame too long: 43>12
[ERROR] [1622190976.973679264]: Exception thrown while processing service call: Ads operation failed with error code 1797.
2021-05-28T10:36:20+0200 Warning: Frame too long: 43>12
[ERROR] [1622190980.381327620]: Exception thrown while processing service call: Ads operation failed with error code 1797.

## I didn't trigger this
2021-05-28T10:39:25+0200 Warning: Port 0x2710 is out of range
2021-05-28T10:39:25+0200 Warning: No response pending
2021-05-28T10:39:57+0200 Warning: Port 0x2710 is out of range
2021-05-28T10:39:57+0200 Warning: No response pending

## 3x reading
2021-05-28T10:51:39+0200 Warning: Frame too long: 43>12
[ERROR] [1622191899.321365716]: Exception thrown while processing service call: Ads operation failed with error code 1797.

2021-05-28T10:51:55+0200 Warning: Frame too long: 43>12
[ERROR] [1622191915.948053277]: Exception thrown while processing service call: Ads operation failed with error code 1797.

2021-05-28T10:58:01+0200 Warning: Frame too long: 43>12
[ERROR] [1622192281.201206366]: Exception thrown while processing service call: Ads operation failed with error code 1797.
```

Having a look with wireshark as suggested in other issues I can see very clearly `ErrorCode: TARGET PORT NOT FOUND (0x00000006)`

Now wouldn't it be great if this text would be printed out somehow? Also, what to do about this error?  I downloaded TwinCat3, created an example project and imported a GVL given by our customer. I assume this should then be AMSPORT_R0_PLC_TC3 right?

(I'm running twincat3 on a windows7 pc and my code on ubuntu 18.04 on a separate pc)

simonschmeisser, 2021-05-28:
Sidenote: I'm super willing to implement the improvements I'm talking about myself, will be in offline in June however

pbruenn, 2021-05-28:
> Now wouldn't it be great if this text would be printed out somehow?
Absolutely! Currently, I am working on something else, but please keep this issue open until we show a better error message.
Regarding, why you see this error in the first place: Is your PLC in RUN mode?

simonschmeisser, 2021-05-28:
Nice! I think there was actually no PLC with that port running (you can see the port in the settings window of the PLC (e.g. Untitled1)), I now re-started with a full example project and it has indeed a PLC at port 851. But I didn't know that PLCs could have a different port

Now I'm seeing error 1827 which pyads describes as `ADSError: access denied (1827)`, when executing the same command directly from the windows pc with pyads I get `symbol not found (1808)` however?? ( The GVL has the attribute `qualified_only` set and therefore I'm trying to read a variable as `GVL._ui_i1_Mikado_ErrorCode`)

pbruenn, 2021-06-02:
Fixed by https://github.com/Beckhoff/ADS/commit/9502174400238a7207ce57ad704edc703c14579e

closed 2021-06-02 by pbruenn

simonschmeisser, 2021-06-02:
Thanks :+1: