Beckhoff/ADS #148: ADSError: access denied (1827)

simonschmeisser, 2021-06-30:
Hi Patrick and others,

Unfortunately I'm still struggeling with reading even a single variable from ADS

I have TwinCat3 running on a Windows 7 PC (at 10.2.0.216 with ID 10.2.0.216.1.1) and this library on an Ubuntu 18.04 with IP 10.2.0.75  and ID 10.2.0.75.1.1 . On the windows PC I added a route back to the linux box
[image]

The code seems to connect successfully but when I then try to read a variable I always get error 1827 (access denied) no matter whether the variable exists or not.

From pyads on the same computer as the twincat I can successfully read and write the variable

My Code:
```
bool AdsIo::tryConnect()
{
    static const AmsNetId remoteNetId { 10, 2, 0, 216, 1, 1 };
    static const char remoteIpV4[] = "10.2.0.216";

    // uncomment and adjust if automatic AmsNetId deduction is not working as expected
    AdsSetLocalAddress({10, 2, 0, 75, 1, 1});

    try {
        m_route = std::make_unique<AdsDevice>(remoteIpV4, remoteNetId, AMSPORT_R0_PLC_TC3);
        initialised = true;
        connected = true;
    } catch (AdsException& e) {
        ROS_ERROR_STREAM("Connection failed: " << e.what());
    }

    return connected;
}

// and for reading

    AdsVariable<uint8_t> readVar {*m_route.get(), "i1_variables_Mikado._ui_i1_Mikado_ErrorCode"};

    std::cout << __FUNCTION__ << "():\n";
    for (size_t i = 0; i < 8; ++i) {
        std::cout << "ADS read " << std::hex << (uint32_t)readVar << '\n';
    }

```

The TwinCAT Static Routes window does not show my connection as connected while the code above `tryConnect()` succeeds

Please let me know how to further debug this.

pbruenn, 2021-06-30:
Just an idea, does pyads automatically prepend “MAIN.” to plc variable names?

simonschmeisser, 2021-07-01:
I don't think it does but I also tested adding "MAIN." to the variable name on C++/linux but it didn't change anything. With pyads/localhost I also get a different error message/code when I try to access a variable that does not exist

simonschmeisser, 2021-07-12:
Hi @pbruenn 
I tried asking the Beckhoff support for further help but haven't received anything helpful yet unfortunately. Could you ask internally when this error code happens and what it could mean?

Thanks a lot!
Simon

pbruenn, 2021-07-14:
Looks like this https://github.com/Beckhoff/ADS/issues/136. And II have absoluty no idea whats going on there. Just like my colleagues, we weren't able to reproduce such behavior. Strange is that pyads works, where our lib doesn't...

simonschmeisser, 2021-07-14:
Hmm, that's unfortunate, I hoped that you have access to whatever code spits out that error code

To clarify: pyads works when running locally on the same windows machine as the twincat software. It shows the same error message 1827 when running from my linux machine. We also tested this with a linux pc connecting to a beckhoff plc at our customer and get the same error there. They have a windows based HMI running just fine with that PLC

simonschmeisser, 2021-07-15:
Our partners did some further testing and found this library to work just fine with TwinCat 2 but give Error 1827 with TwinCat 3, maybe that helps? They also captured some data with wireshark, I just asked for permission to share that with you.

simonschmeisser, 2021-07-19:
It turns out that one should *not* set the `unidirectional` checkbox in the route setup dialog on the host/windows plc

I'll see if there is a good place for this (and also a hint for not using secure ADS) in the readme

closed 2021-07-19 by pbruenn

simonschmeisser, 2021-07-19:
Thanks for your help @pbruenn both here and in direct communication!

Dominic-Liu, 2023-11-24:
Hi Team,
appreciate your support.
i got a error while using ads node, the result is that: failed to connect 192.168.193.10.1.1:851(Twincat AMS Net Id: in my VM windows). detail please refer to below:
PC IP: 192.168.193.1,
VM windows IP:193.168.193.10, i installed the twincat3 in VM windows.
PC connected with VM windows via virtual bridge.
[image]
[image]
[image]
[image]
look forward you support. Thx..

PR #149 (2021-07-19): More details about route configuration
fixes #148
commit 82c281d7 (2021-07-19): More details about route configuration

fixes #148