Beckhoff/ADS #150: Router CleanUp

RamJamGit, 2021-09-09:
Unfortunately, the Router->CleanUp function cannot be called via a high-level language (C++, C#) interface. Is there a possibility to initiate the Router->CleanUp via ADS?

[image]
[image]

RamJamGit, 2021-09-14:
@soberschmidt The Router->CleanUp should be executed locally and the tray menu, which contains the menu item, is deactivated. In the case of terminated processes (crash) under TwinCAT 2, the CleanUp is used to close the ADS connections that can no longer be used by the router. As a result, the crashed process (zombie process) is separated from TwinCAT and can then also be terminated.

soberschmidt, 2021-09-14:
@RamJamGit This issue is not related to the open source ads client of this repo. Please contact support@beckhoff.com

closed 2021-09-14 by soberschmidt