Beckhoff/ADS #159: How does the `adstool` work?

FF-AntiK, 2022-02-17:
I have two devices. One Development PC running Linux and an IPC running TCBSD.
On SSH-Login the IPC shows:
```
TcOsSys.dll:	TcOsSys_Rel31_4024_20211129.2
TwinCAT Build:	3.1.4024.22
AMS Net Id:	5.65.251.130.1.1
TC/BSD:		12.2.12.0,2
```

When I try to run `adstool 5.65.251.130.1.1 --log-level=0 state` I get the following result:
```
2022-02-17T15:41:53+0100 Error: AdsException message: Ads operation failed with error code 6.
```

What am I missing?

pbruenn, 2022-02-21:
By default, only ads-s (ADS secure) is enabled in the TC/BSD firewall. But the AdsLib and the adstool support only classic ADS. So either forward the classic ADS port (48898) over something secure like ssh. Or enable ADS in your TC/BSD firewall by adding `pass in quick proto tcp to port 48898` to `/etc/pf.conf` and reload is e.g. with `doas pfctl -f /etc/pf.conf`

FF-AntiK, 2022-02-21:
Thanks for your reply! I've stopped `pf` now completely but I still get the same error message. Do I have to setup a `route` first? How would I do this?

pbruenn, 2022-02-21:
Yes, you need a route. From your Linux machine: `adstool <tcbsd_ip> addroute --addr=<linux_ip> --netid=<linux_netid> --password=<tcbsd_password>`

For linux_netid you can choose anything which is different from tcbsds netid. for subsequent call you should then use `adstool 5.65.251.130.1.1 --localams=<linux_netid> state`

FF-AntiK, 2022-02-21:
I've added the route which seems to work (at least there was no error code).
But reading the state still returns with Error Code 6.
But now it's not printing the error, even with `--log-level=0`

pbruenn, 2022-02-21:
Assuming:
- your Linux has the ip 10.0.200.1
- your TC/BSD has the ip 10.0.200.10
- your TC/BSD password is the incredible "1"
- you choose 1.2.3.4.5.6 as your Linux AmsNetId
- you TC/BSD allows TCP connections to port 48898
- TcSystemService is running on TC/BSD:
 ```
Administrator@PC-8CD98F:~ $ doas service TcSystemService status
TcSystemService is running as pid 789.
```

Then this should work:
```
adstool 10.0.200.10 addroute --netid=1.2.3.4.5.6 --addr=10.0.200.1 --password=1
adstool 5.65.251.130.1.1 --gw=10.0.200.10 --localams=1.2.3.4.5.6 state
```

FF-AntiK, 2022-02-21:
`--gw` did the trick!
Thank you very much!

closed 2022-02-21 by FF-AntiK